ExpressLRS/ExpressLRS #2468: Elrs Configurator 1.6.1 does not build Frsky R9M Lite Pro in manual mode with elrs 3.3.1 - r9m_lite_pro_bootloader.bin "not found" error

alccrl, 2023-11-05:
[loglog-3.3.0-1.6.txt](https://github.com/ExpressLRS/ExpressLRS-Configurator/files/13260381/loglog-3.3.0-1.6.txt)
[loglog-3.3.1-1.6.1.txt](https://github.com/ExpressLRS/ExpressLRS-Configurator/files/13260382/loglog-3.3.1-1.6.1.txt)
[image]
Configurator 1.6 complains a 3.3.1 file or the like (see screenshot).
Configurator 1.6.1 does not flash (see log-3.3.1-1.6.0 file), bootloader bin is not found (actually it exists).
Configurator 1.6 succedes in flashing 3.3.0 (see log-3.3.0-1.6 file).

I guess some issue in python scripts or cloud structure 3.3.1 against compilation / flashing local files structure of 3.3.0,
problem persists also after deleting all configuration and files before installing different versions.

Platform:
Linux Ubuntu 23.04
St-link v2
Elrs 3.3.0 / 3.3.1
Configurator : 1.6 / 1.6.1 (both have problems, se description above)
Hw : Frsy R9M Lite Pro in st-link mode

Alfredo

simmis, 2023-11-09:
I am experiencing this exact same bug (ELRS 3.3.1 & Configurator 1.6.1) on the same OS Ubuntu 23.04. 
The bootloader file exists. 
```$ find . -name r9m_lite_pro_bootloader.bin
./firmwares/cloud/ExpressLRS/ae9df3d202aa905358f0fdc1ca36718736ebdf38/firmware/bootloader/r9m_lite_pro_bootloader.bin
./firmwares/cloud/ExpressLRS/e051b89b23b5935b959ea9beddf685d857d03af9/firmware/bootloader/r9m_lite_pro_bootloader.bin
```
But the script seems not to be looking in the right place. Same `FileNotFoundError` as reported above.

simmis, 2023-11-10:
For a workaround see here: https://discord.com/channels/596350022191415318/1172484941045563423

closed 2024-04-06 by pkendall64