ExpressLRS/ExpressLRS #2471: Change WiFi targets firmware binaries extension from .bin.gz to .bin

jurgelenas, 2023-11-07:
Users are extracting the compressed firmware binary before they upload it through the WiFi update page, and it results in "Not Enough Space" error.

Firmware binary files compression is not important implementation detail. We could retain the usual `.bin` file extension to prevent issues in the future.

wilderk, 2023-11-10:
I was one of the "smart" people that decompressed the bin.gz firmware file before flashing and got the "Not Enough Space" error. Now I know better.
An informative note in the Configurator explaining that flash by wifi creates a bin.gz file is another option. 
I think it is important to know that the file is compressed with bin.gz file extension so that someone does not try to flash a wifi compressed firmware file by another method like UART or BFpassthrough.

0crap, 2023-11-20:
Changing this now will make the confusion even bigger.
But I do agree that it's better to change back to .bin on the long run.
(Support will not be happy with this.)