ExpressLRS/ExpressLRS #2487: How to join

tobeornottobe81, 2023-11-25:
Due to network issues, we can only ask questions here.

We want to add remote control logos with our own characteristics to this project. What conditions need to be met? Thank you.

YI-BOYANG, 2023-11-25:
你是想添加飞盈佳乐的开机logo?

pkendall64, 2023-11-25:
> Due to network issues, we can only ask questions here.
> 
> We want to add remote control logos with our own characteristics to this project. What conditions need to be met? Thank you.

You need to get in contact with one of the Devs on discord. Then we can talk about how you go about becoming an approved manufacturer to get your targets included the firmware and configurator.

tobeornottobe81, 2023-11-27:
> 你是想添加飞盈佳乐的开机logo?
 谢谢回复，我们设计了一套接收机硬件，想要客户在使用ExpressLRS Configurationr的时候能够在目标中选择以我们公司命名的设备。

tobeornottobe81, 2023-11-27:
> You need to get in contact with one of the Devs on discord. Then we can talk about how you go about becoming an approved manufacturer to get your targets included the firmware and configurator.

thank you，that's our idea.

closed 2023-11-29 by tobeornottobe81