ExpressLRS/ExpressLRS #2490: 10kHz PWM output does not go to 0% or 100% Duty on ESP8266 based receivers

ot0tot, 2023-11-28:
I verified the behavior described below as well. 10kHz output works properly on ESP32 based receivers, but not on ESP8266 based receivers. Output initializes properly at zero, but once changed never goes below about 1% or above 99%. 

Original report:
I can now confirm that my Channel goes from 988-1884. Tested different elrs Versions and differed RX pins. 

The very strange thing is that the pwm signal on the receiver Pin is 0 (without any spikes) when the receiver is powered up and bound with the TX while the TX channel signal is low. As soon as the channel signal was above 1000µs once, the receiver output pwm will never go back to o again, it will always keep small positive spikes making to motor to spin slowly and never stop. 
For me this looks like a bug in the pwm generation, anyone can confirm this?  maybe @Ncerzzk ?

Feel free to contact me in Discord: https://discord.com/channels/596350022191415318/798006228450017290/1168851700002590751

_Originally posted by @derFrickler in https://github.com/ExpressLRS/ExpressLRS/issues/1850#issuecomment-1786970968_

derFrickler, 2023-11-28:
As i read that you were using a BetaFPV Lite RX as well, i just crosschecked with an other Happymodel EP1 and the behavior is exactly the same. Seems to be on all ESP8285 RX.

ot0tot, 2023-11-28:
@CapnBry Does anything jump out at you as possible causes? Maybe a timer/interrupt issue on ESP8266 with https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lib/PWM/waveform_8266.cpp ?

I suppose the trivial solution is just to add a "if duty =0, digitalWrite(pin, LOW); if duty = 100, digitalWrite(pin, HIGH);"

PR #2496 (2023-12-18): Fix for 10KHz 0% and 100% on 8285 receivers
Would you folk like to test this please and comment here if it fixes it for you. @ot0tot @derFrickler 

Fixes #2490
closed 2024-01-18 by ot0tot