ExpressLRS/ExpressLRS #2503: Min/dyn power at Jumper T20 with AION T-20 900M TX

sandrosov85, 2023-12-11:
Hello guys! 

With 3.3.1 FW and this [verison](https://github.com/EdgeTX/edgetx.github.io/blob/master/downloads/T20-PROD-09-07-23.bin?raw=true) of EDGE TX in setup of ELRS module i have only 250 mW on ELRSv3.lua. No matter dyn=on or off.

Backplate becomes heat over 5 minutes and i think HW module working not normal (maybe in max power). 

Can you check sync settings HW module and FW setup? 
It is would be great if options of 25mw tx power becomes accessible.

I'll be happy to help in any way I can.

- TX hardware: AION T-20 900M TX
- RX hardware: FRSKY R9 Slim Plus
- Handset model: Jumper T20
- OpenTX version (including nightly number): T20-PROD-09-07-23
- ExpressLRS version (TX & RX MUST MATCH): 3.3.1
- Packet Rate: 200
- Telemetry Ratio: 1/8
- user_defines:

pkendall64, 2023-12-11:
It's a limitation of the hardware. 250mW is the minimum it can go to.

closed 2023-12-11 by pkendall64

sandrosov85, 2023-12-12:
Very strange. I buy handset with ELRS 2.2.5 instaled on AION TX and option of 25mw was available with ELRSv3.lua.

pkendall64, 2023-12-12:
Indeed, as the Jumper T-20 900 TX was only added in 3.3.0. The 2.4GHz module does go down to 25mW though and was also added in 3.3.0.

The T-Pro and the AION Nano module were available in 2.5.2. These are 2.4GHz modules and also go down to 25mW.