ExpressLRS/ExpressLRS #2512: RX wifi update issue

JarrodB92, 2023-12-22:
## Current Behavior
With both Radiomaster RP1 and Happymodel EP1 RX led is now dim after wifi update to 3.3.1  I built the update and flashed via wifi and now both receivers do not fully light up and do not connect at all to the TX.

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware: RadioMaster Ranger
- RX hardware:Happymodel EP1 & Radiomaster RP1
- Handset model:RadioMaster TX16s
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): 3.3.1
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2023-12-22:
Please as on discord in the #help-and-support channel and I'm sure the friendly folk there will help you out.

JarrodB92, 2023-12-22:
Hi Paul, I was able to solder the boot pads and perform a betaflight
passthrough and successfully update them. I’ve just never seen the issue
that I posted happen.

pkendall64, 2024-01-02:
This issue happens every now and then when the receiver is re-powered before the LED has resumed normal blinking pattern after a flash.

closed 2024-01-02 by pkendall64