ExpressLRS/ExpressLRS #2524: Telemetry lost followed by loss of control ELRS 3.3.1

shaun190, 2024-01-05:
Radiomaster TX16s Mk1
OpenTx 2.3.15 (1004ef7c)
beta fpv 1w module (Elrs 3.3.1)
Radiomaster er8g receiver EU LBT (Elrs 3.3.1)
Fixed wing Radian foamy glider with 6 servos and esc motor control

Flights conducted in open airfield area all LOS and at maximum range of 500m from tx.  Nearby comms mast and other airfield infrastructure.

Two flights with no problems but swamping? with multiple telemetry lost announcements at close range during launching.
Third flight gave telemetry lost announcement followed by immediate loss of connection to the receiver and model.  
Model landed safely in free flight.  
After recovery the only way to establish connection was to turn off the receiver and turn on again.

Is this the previously mentioned bug in earlier EU LBT ELRS versions returned again?

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

shaun190, 2024-01-05:
Need to add:
TLM 1:8
Packet rate 100hz  full

m0p3d, 2024-01-13:
Without any data e.g. log´s nobody can check the situation.

shaun190, 2024-01-13:
Thanks for the feedback. This is my first venture into using the elrs protocol.
I wondered if anyone else had had a similar problem with this version.
I thought I had set up the logs but unfortunately nothing recorded on the SD card.
I was also surprised with the quantity of telemetry lost warnings when the tx was close to the model.  Is that normal with Elrs?

m0p3d, 2024-01-15:
Its not typical. Very close rx,tx distance can result into drop of link quality (<1m)

shaun190, 2024-01-15:
Thanks for the information.
I have tried the 240LU setting and also the “lock on last” one too but it didn’t seem to make much difference.
I did try a test with a Betafpv micro PWM receiver in a different glider at the same time as the Radiomaster one I had the problem with and this didn’t seem to have the telemetry lost messages even when the tx and rx were close to each other (<1m).  I also had 3 flights with this at the same time but no loss of signal and no telemetry lost message so not the air either.

shaun190, 2024-01-18:
Just had an opportunity to play with this again.
Tried the original setup and got a load of telemetry lost messages as before and also  jittery servos particularly when not armed.
 I then loaded ELRS 3.3.2 to the tx and also the ER8G rx.
I also changed the telemetry setting to 360LU.
The telemetry lost warnings have now all but disappeared unless I put the tx on top of the rx and the connection seems to also produce less jittery servos than before.
I need to test a little further and also try flying with this new firmware.
Not very scientific I guess but it’s the best I can manage at the moment.

shaun190, 2024-01-29:
Have now flown this new set up with the Radiomaster ER8G receiver and put in over an hours air time all VLOS flying but in the area where I have always previously experienced multiple telemetry lost messages and also where I had the free flight incident.
No messages in all the flying time and a similar result flying a Betafpv micro PWM receiver also running on ELRS 3.3.2.
It is now possible to get the tx to within less than 100mm of the rx and have no telemetry lost messages on the bench.  Looks like 3.3.2 has solved the problem.
I also tried a Betafpv Moxon antenna which seemed to improve the RSSI and signal quality for me.

closed 2024-01-30 by pkendall64