ExpressLRS/ExpressLRS #2531: Sporadic channel value changes

KissUltra, 2024-01-12:
## Current Behavior

Few times per flight, channel data (throttle and pitch) are jumping high for 1 erls packet.

## Steps to Reproduce

Watch logs. changing refresh rate from 500hz to 250hz makes it just twice longer as expected.

[image]

KissUltra, 2024-01-12:
[image]

KissUltra, 2024-01-12:
[image]

AlinVictor, 2024-01-12:
[image]
[image]
[image]
[image]
[image]

Tx-Radiomaster Boxer with internal ELRS module. (V3.2.0), EdgeTX 2.8.4
Rx-HappyModel EP1 Dual TCXO (V3.2.0)

SupaflyFPV, 2024-01-17:
Have you tried or seen this issue on another radio/gimbal? If it was there we would have seen it at Betaflight....

SupaflyFPV, 2024-01-17:
> ![5](https://private-user-images.githubusercontent.com/66276171/296160432-2131ecfc-835f-479a-893e-27fc7a71208a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MTg1OTEsIm5iZiI6MTcwNTUxODI5MSwicGF0aCI6Ii82NjI3NjE3MS8yOTYxNjA0MzItMjEzMWVjZmMtODM1Zi00NzlhLTg5M2UtMjdmYzdhNzEyMDhhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDE5MDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIxZjZiODQwNzI1M2NlNjdiMGUzMWU5ZmQxNmQyZGIyMDk2ZmZmMjY4OTAwYjgzODc0MGIxOWEzMDQ5MzRjYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PJtvDpa-oB07L-EQ2vU-kvQMQ92c-aWcZoEvEoS3mvA) ![500hz4](https://private-user-images.githubusercontent.com/66276171/296160441-0c518abd-2715-4de4-a49b-7d4819a6c86c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MTg1OTEsIm5iZiI6MTcwNTUxODI5MSwicGF0aCI6Ii82NjI3NjE3MS8yOTYxNjA0NDEtMGM1MThhYmQtMjcxNS00ZGU0LWE0OWItN2Q0ODE5YTZjODZjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDE5MDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNmMmJhYzhmYzYwZjU5MzI1OTlmNWY4ZDdiMGM3ZmZmZTlhMjJmMjk5MGI0OThkOWUwZjE0ZjRiN2ViY2ZkY2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.xVQfI__VnCAQUPGZPK2YFUM0ELIEyHqAjc1kM04cOsw) ![2501](https://private-user-images.githubusercontent.com/66276171/296160444-f8eaa72d-d3d9-49d4-9853-4cca1508c198.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MTg1OTEsIm5iZiI6MTcwNTUxODI5MSwicGF0aCI6Ii82NjI3NjE3MS8yOTYxNjA0NDQtZjhlYWE3MmQtZDNkOS00OWQ0LTk4NTMtNGNjYTE1MDhjMTk4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDE5MDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1NDdhMTFlMzZlMTc5M2FkYzg1NTBlYzg5ZmQ4MTc4YTI4YjlmNTM0ODRmNDJiZWRlMDc0M2NhNWIxNzFiN2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.p1HucMscGWv-dZcu6c7cNnxflxkepDdiZV9eA9V8aDw) ![2502](https://private-user-images.githubusercontent.com/66276171/296160447-252e85c2-a96c-4891-84ce-e5aada56c36b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MTg1OTEsIm5iZiI6MTcwNTUxODI5MSwicGF0aCI6Ii82NjI3NjE3MS8yOTYxNjA0NDctMjUyZTg1YzItYTk2Yy00ODkxLTg0Y2UtZTVhYWRhNTZjMzZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDE5MDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFkMjVjY2U5OTdhMTk2MzZmY2Y4ZmQxN2ExMTIwNWYxZDFlNTliYzFiMDU5MzY1NjAyM2ZiYzc5NGJjYWU5OTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Pe9HQOw4r4cxewyCd_8-Z5h1IMeUpL8T-p5RpZC6D5I) ![5003](https://private-user-images.githubusercontent.com/66276171/296160450-cf450d8e-2937-4ab1-b04f-11e71b373f22.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MTg1OTEsIm5iZiI6MTcwNTUxODI5MSwicGF0aCI6Ii82NjI3NjE3MS8yOTYxNjA0NTAtY2Y0NTBkOGUtMjkzNy00YWIxLWIwNGYtMTFlNzFiMzczZjIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDE5MDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM4OGJhNDJhMGY0MDU4M2JiYTY3MzRkMDNkZjUwNWVlYTkzZGE2YWFiNzNlOTRjOTczYTE0YTc3YmNlMzM1NTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lbSAzKSWO6s-Zx6WY5y_bDC-o1rSmeR4NClKmz5IGKA)
> 
> Tx-Radiomaster Boxer with internal ELRS module. (V3.2.0), EdgeTX 2.8.4 Rx-HappyModel EP1 Dual TCXO (V3.2.0) On the same radio, with a Crossfire MicroTX module, this issue doesent happen. I even mount two receivers on a quad(elrs and crossfire) and the issue happen only with ELRS. Thanks!

out of interest have you tried another ELRS receiver, and how many other cases are there out there in Kiss world where this is happening? I didn't see it in BF however it may be flying under the radar...

KissUltra, 2024-01-17:
> 
> Have you tried or seen this issue on another radio/gimbal? If it was there we would have seen it at Betaflight….
> 

You have to see RAW input from the receiver. Without ANY filtering.

JyeSmith, 2024-01-17:
@KissUltra @AlinVictor can you please let me know the FCs being used, firmware version and baud settings.

KissUltra, 2024-01-18:
Ultra. Any firmware. You can see other channel data coming OK, and since its crossfire packet (one) its protected with checksum. Baudrate is irrelevant imho. Once more. I DO NOT filter rc input, so its RAW data from receiver.

JyeSmith, 2024-01-18:
@KissUltra  I think baud is important here because there has historically been some variation in the used crsf baud between projects.  Any differences may cause issues with reading the serial packet.  Yes the packet has an 8 bit crc, but it isnt a very strong crc for the packet length.  So baud issues that may result in flipped bits could cause a problem here.  So I would like to rule this out, and also an FYI that the ELRS baud can be changed in the UI to match.

KissUltra, 2024-01-18:
@AlinVictor is testing atm.

JyeSmith, 2024-01-18:
> Well, its quite good crc. Lets say good enough to detect such change. Anyhow, we are testing on right baudrate. H7 has quite a tolerance on baudrate, so never had issues with crossfire really.

ELRS uses 420k as the original devs used BF and matched the bauds.

What baud does KISS use?

Have you adjusted the baud in the ELRS UI?

KissUltra, 2024-01-18:
historically i use 400000. Which is well within tolerances of 416khz of tbs.

KissUltra, 2024-01-18:
yeah, guys will try that. 420000 can be just on the edge of tolerances.

SupaflyFPV, 2024-01-18:
Updates looking better with baud 400 on rx and radio - (from 420k and 1.87Mln) - good call @JyeSmith 

Before
[image]
[image]

After

[image]

KissUltra, 2024-01-18:
Yes, i am sticking to v06 crossfire protocol for backward compatibility. its 400kbaud. It seems indeed changing speed to 416 or 400 (makes no difference) improves that. I also added to osd amount of crc errors detected. @AlinVictor making list with results of the testing. Its pretty annoying to ask people now to change baudrate using erls. If it at least be kept to standard - it would JUST WORK for BF and KISS. 

Once more. CSRF 0.6 defines 400kbaud. CSRF 0.7 defines 416kbaud. I think its related to hardware tbs used. Anyhow, advice of trappy to stick to 400kbaut seems just works for crossfire/tracer just fine. I am really puzzled where 420kbaud number came from. Its just WRONG. 
[image]
[image]

For ultra users, next version will have statistics for rx crc errors

[image]

JyeSmith, 2024-01-18:
> Thanks for your help and time to assist Kiss users @JyeSmith 

You're Welcome.

> Once more. CSRF 0.6 defines 400kbaud. CSRF 0.7 defines 416kbaud. I think its related to hardware tbs used. Anyhow, advice of trappy to stick to 400kbaut seems just works for crossfire/tracer just fine. I am really puzzled where 420kbaud number came from. Its just WRONG. And ERLS just violates standard it uses. Stick to the crossfire documents guys if you claim using crossfire protocol. They are here for a reason. No need to reinvent the wheel, its already done by TBS years ago. :)

The baud issue is a known one, but 420k is common across ELRS, BF, INav, Ardu and probably more.  AFAIK TBS contributes to some of these and could correct if they want.

Maybe you could share those documents on this issue and help the community in sticking to the standard?

https://github.com/tbs-fpv/freedomtx/issues/26

ELRS has had setup information for a number of years to correct the baud for kiss users.

[image]

closed 2024-01-18 by JyeSmith

brandonrc, 2024-01-19:
@KissUltra puzzled where 420kbaud came from? I hope you are not the one that made Kissultra... 

https://github.com/betaflight/betaflight/blob/42267349db0c92a0e1063bd1d07633a3322dfb07/src/main/rx/crsf_protocol.h#L30

https://github.com/iNavFlight/inav/blob/1d12c7f3dcf410ce600fdf2ae8e37677b062aa07/src/main/rx/crsf.h#L20

https://github.com/ArduPilot/ardupilot/blob/6ce4dfea5747e959746f652fe823c7e7458693bb/libraries/AP_RCProtocol/AP_RCProtocol_CRSF.cpp#L63

> Once more. CSRF 0.6 defines 400kbaud. CSRF 0.7 defines 416kbaud. I think its related to hardware tbs used. Anyhow, advice of trappy to stick to 400kbaut seems just works for crossfire/tracer just [...]

ot0tot, 2024-01-19:
https://github.com/ExpressLRS/ExpressLRS/issues/2531#issuecomment-1898725453
> Updates looking better with baud 400 on rx and radio - (from 420k and 1.87Mln) - good call JyeSmith

No need to change the baud rate on the radio side, it has nothing to do with the receiver baud rate. Leave it at 1.87M or whatever is appropriate for your radio. 
https://www.expresslrs.org/quick-start/transmitters/tx-prep/#serial-baud-rate

KissUltra, 2024-01-19:
> @KissUltra puzzled where 420kbaud came from? I hope you are not the one that made Kissultra... [...]

Thanks for info. BUT. Same file and comment... From ARDU.

[image]

KissUltra, 2024-01-19:
> > Thanks for your help and time to assist Kiss users @JyeSmith
> 
> You're Welcome.
> 
> > Once more. CSRF 0.6 defines 400kbaud. CSRF 0.7 defines 416kbaud. I think its related to hardware tbs used. Anyhow, advice of trappy to stick to 400kbaut seems just works for crossfire/tracer just fine. I am really puzzled where 420kbaud number came from. Its just WRONG. And ERLS just violates standard it uses. Stick to the crossfire documents guys if you claim using crossfire protocol. They are here for a reason. No need to reinvent the wheel, its already done by TBS years ago. :)
> 
> The baud issue is a known one, but 420k is common across ELRS, BF, INav, Ardu and probably more. AFAIK TBS contributes to some of these and could correct if they want.
> 
> Maybe you could share those documents on this issue and help the community in sticking to the standard?
> 
> [tbs-fpv/freedomtx#26](https://github.com/tbs-fpv/freedomtx/issues/26)
> 
> ELRS has had setup information for a number of years to correct the baud for kiss users.
> 
> ![image](https://private-user-images.githubusercontent.com/14170229/297876465-f5617802-687c-4c58-a7e2-3e7b43e328fd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2NDUzOTMsIm5iZiI6MTcwNTY0NTA5MywicGF0aCI6Ii8xNDE3MDIyOS8yOTc4NzY0NjUtZjU2MTc4MDItNjg3Yy00YzU4LWE3ZTItM2U3YjQzZTMyOGZkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE5VDA2MTgxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVlZmU4YzFhNGI0OGVhZGIyMjFkZTQxZGFjYWZkYjg3ODEyMDE4MGI3NjMyYWI5NzdhZDIyOGE1NzBjNzE1YjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.lRFxuJ9vyaZImBzRQ1GqmVi8meGeLFdWWfMjUzjm2wA)

I cant share full document without permission of TBS because i am under NDA. I asked for it.

KissUltra, 2024-01-19:
Also, from freedomtx git discussion:

[image]

JyeSmith, 2024-01-19:
> Also, from freedomtx git discussion:
> 
> <img alt="Screenshot 2024-01-19 at 07 40 45" width="956" src="https://private-user-images.githubusercontent.com/86871477/297975856-ad7f74ef-f5cd-4e62-a9a8-7dabfaba8a16.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2NDcxOTIsIm5iZiI6MTcwNTY0Njg5MiwicGF0aCI6Ii84Njg3MTQ3Ny8yOTc5NzU4NTYtYWQ3Zjc0ZWYtZjVjZC00ZTYyLWE5YTgtN2RhYmZhYmE4YTE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE5VDA2NDgxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNmYWNmZDUxMmYxN2JkMDhiZDI1Y2JmMjU1ODVjOGY4OWM5Y2Q0OTRjNjc5Y2NjZDhkYzdmZTIxMTQ1OGIzMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pUfVJuCi48T_HVFgmKuJaBONGV3-wf1PoFY1Sjy2WSc">

Ah cool.  Updating kissultra sounds sensible and will support your users 👍

KissUltra, 2024-01-19:
i think this discussion becomes pointless. You don't listen when you told what you have to use. I rather suggest to fiddle with receiver config or use crossfire. Thank you for nothing. Case closed.

KissUltra, 2024-01-19:
Only reason i raised this question because you guys doing something off standard. Sorry, just wanted to help YOUR product. I guess its not needed improvement then.

radio-mistress, 2024-01-19:
Let me see if I got this right. 
- issue opened with known baud rate mismatch
- customization is possible and already documented
- you then link what you you should be doing which is updating the flight control software (like the ardu example) to handle ELRS 
[image]

- instead, you expect ELRS to break its existing compatibility out of box the with countless released versions of open-source software so it can maintain compatibility with a closed-sourced codebase and better match internal TBS documentation instead of the open documentation https://github.com/crsf-wg/crsf/wiki/Receiver-Baud-Negotiation

- surprisedpikachu.jpg when the conversation goes nowhere.

KissUltra, 2024-01-19:
Sure thing. Bye.

SupaflyFPV, 2024-01-19:
> i think this discussion becomes pointless. You don't listen when you told what you have to use. I rather suggest to fiddle with receiver config or use crossfire. Thank you for nothing. Case closed.

Put the 400k baud rate ELRS recommendation into Kiss Ultra wiki - job done 😎