ExpressLRS/ExpressLRS #2541: Can't save binding phrase using the latest master

flux242, 2024-01-23:
Ive just synced to the commit 212361bb015af8dca57065333c7d290b85ff5540 and build the TX FW. After flashing I can connect to the WiFi web form and I can type a new binding phase in but the 'Save' button stays inactive. Why is that?

pkendall64, 2024-01-29:
This generally happens when the page has not fully loaded i.e. the javascript has not loaded completely.

flux242, 2024-01-30:
ok, I wanted to check js network communication in the browser and flashed the fw again. This time I didnt have any problem and everything works as expected. I'm not sure what I did differently previously but my suspect is the build process that asks questions now at the end. Maybe I typed something wrong at that step. But I remember having a 'Buttons' tab in the gui previously but not now.

closed 2024-01-30 by flux242