ExpressLRS/ExpressLRS #2563: Incorrect Model ID using Model Match should not bind or create any connection at all

christhomas, 2024-02-17:
I have multiple drones. I set them up like this
- Goose with Model Match ID 1
- Cyclops with Model Match ID 2

Then I setup both my Radios Zorro and BetaFPV LiteRadio 3 Pro with two models
- Goose with Receiver ID 1
- Cyclops with Receiver ID 2

I power on the Zorro, select the Goose Model (ID 1)
I power on Cyclops programmed with ID 2
The Zorro connects to Cyclops ¯\_(ツ)_/¯ 

I power on the LiteRadio 3 Pro, select the Cyclops model (ID 2)
The LiteRadio 3 does not connect, won't connect

The Zorro still shows it's connected to the Cyclops model, despite not having the correct model id

I think the drone should disconnect from the radio when the id model is wrong. According to the documentation, it does not create a full connection when the model id is wrong. I think this needs to change to disconnecting from the radio and going back to search for a new connection.

I am not entirely sure what the point of model match is, if it's not used to match a model with a drone (or wing, plane, etc)

pkendall64, 2024-02-17:
When bound using the same bind phrase the drone will connect to the TX with a different model id. With model match enabled, it will still connect but the RX will not send any data to the FC or drive PWM outputs. In the Lua script you will see "Mismatch" on the top status line.

For more information see the documentation https://www.expresslrs.org/software/model-config-match/

closed 2024-02-17 by pkendall64

christhomas, 2024-02-17:
I understand what it does, I'm saying it should disconnect and go back to searching for a connection. It should not remain connected to the TX since it does not give the correct model id

christhomas, 2024-02-17:
The problem is, once it remains connected (bound) to the TX that has the wrong model id, it's impossible to connect it to the correct TX with the correct model id.

Whats the point of remaining connected to a TX that can't do anything, other than remain connected. If the user does this, it should not work, not bind, and the user should then select the correct model in the TX that has the correct model id

christhomas, 2024-02-19:
This ticket is marked as closed, but it's shouldn't be. What should I do now, create a new ticket?

pkendall64, 2024-02-20:
The current implementation is not a bug. It's is designed to work this way.
The reason it semi-connects to a TX is that it has the correct bind phrase and displays "Mismatch" in the Lua script and flashes the LED on the receiver with triple blink to show that the model id is incorrect or the model-match flag is different.

christhomas, 2024-02-20:
Right ok, so instead of saying it's a bug, I'll say it's better to change it so it does not stay connected. The radio already can be told the model didn't match and store that value to show in the lua script later and then you disconnect and ready to accept another attempt to connect from a radio that has the correct binding phrase and correct model id.

I don't think it's useful to stay connected to the radio that has the wrong configuration. You can't disconnect it and you're blocked until you turn off your radio and turning the radio back on means there is a chance to bind to the wrong radio again. Better to send back the "model mismatch" message, disconnect and go back to waiting for a new connection from a different radio.

What do you think?

pkendall64, 2024-02-21:
This has been discussed by the devs when the model match feature was first implemented.

It stays semi-connected so that the user can reset or disable/enable the model match for the current model Id via the Lua script, which will send the model Id/match flag to the connected receiver.

christhomas, 2024-02-21:
Ok, but can I suggest a simpler solution to this problem. Just connect using the right model. That's it.

Why complicate and confuse the situation by staying connected to a radio that has the wrong model loaded just so the user can disconnect model match in the lua script, when the user can just change the model they are using to connect to the correct one.

Cause as it stands right now, this is faulty logic. The wrong model id was used to connect, but instead of just refusing to connect the radio stays connected. In some quasi-grey middle ground instead of having clear and precise logic. Wrong model, refuse to connect. Force the user to correct the problem by selecting the right model.

Doesn't that sound more reasonable, more understandable, less confusing, less problematic, more deterministic?

Wrong model id, send back error response, disconnect and wait until next bind attempt. 

Easy and simple. Do you know the reason why that was not the chosen way to solve this problem? This seems like a problem that doesn't need to exist.

christhomas, 2024-02-26:
So was there any reason why the devs chose to not implement that solution?