ExpressLRS/ExpressLRS #2564: Most packet rates transmit only half as fast, when the LUA script is closed

bastian2001, 2024-02-18:
## Current Behavior

On some setups (to be determined, which ones), ExpressLRS packets are only transmitted half as fast, after the LUA script is closed. Even after a reboot of the remote. I.e. when 50Hz is selected, ELRS transmits at 25Hz when I am on my remote's home screen, jumps to 50Hz, when I enter the LUA script, and goes back to 25Hz when I 

## Steps to Reproduce

1. Open LUA script
2. Set to F1000 if not already done (or any of the other values listed down below)
3. Arm the quad to start blackbox logging
4. Rotate one of the sticks continuously to get a constant stream of new values to look at later
5. While doing that, close the LUA script on the remote
6. Disarm
7. Look at blackbox logs

## Details

- This bug only appears when using telemetry. When telemetry is turned off in the LUA script, the full rate is used all the time. Once a telemetry ratio is selected (even 1:128), the packet rate is halved _as soon as I close the tool_.
- I recommend turning RC smoothing off to better see the issue.
- I first posted this as an issue in [EdgeTX](https://github.com/EdgeTX/edgetx/issues/4642), but since I was told that EdgeTX treats D500 and F500 equally, I concluded this has to be an ELRS issue instead. See that issue for the full initial discussion. I put the TL;DR in this issue too.
- Please find below excerpts of the blackbox logs of my FC as well as the logs from the betaflight FC. Roughly at the middle of the picture, the LUA script was closed. It is visible that before this, the packets come twice as often as after closing the script.
[image]
[image]
- The "magic fix" (i.e. the effect of the bug disappearing because the LUA is open) also happens when opening some of the other LUA scripts on my remote: Model Locator, ExpressLRS, Multi chan namer, Multi LOLI RX config, MultiConfig or TBS Agent Lite
- Armed or not (first AUX channel) does not make a difference
- Every 5s, the packet rate jumps to 1kHz, for a brief moment (up to 0.5 seconds) before it goes down to 500Hz again. More precisely: For around 100-500ms the FC gets one packet every ms, and then for the next 5s, there's only one packet every 2ms. Then there's one packet every 1ms for 100-500ms again, and then for the next 5s, there's only one packet every 2ms.
- This issue does **not** appear on FlySky FS-I6X running OpenI6X (a port of a rather old OpenTX version)

## Your Environment

At home and in the field

- TX hardware: Radiomaster Ranger Micro (tried on two of them)
- RX hardware: Radiomaster RP1 and [DIY RX with TD and PA](https://github.com/bastian2001/Kolibri-FC/blob/main/PCB/PCB%20Version%200.4/Exports/PCB%20Version%200.4.pdf) (similar to [AnyLeaf Dual Radio RX](https://github.com/AnyLeaf/elrs-hardware/blob/main/dual%20radio%202.4Ghz%20schematic.png))
- Handset model: Radiomaster TX16S
- EdgeTX version: 2.9.2 and 2.9.4
- ExpressLRS version (TX & RX MUST MATCH): 3.3.2
- Packet Rate: 50, 100 Full, 150, 250, 333 Full, 500, F500, F1000
- Telemetry Ratio: tested on 1:128, 1:32
- user_defines: none
- Tested on [my own FC](https://github.com/bastian2001/Kolibri-FC) running custom Firmware, and verified on another quad running Betaflight 4.4

mha1, 2024-02-18:
just a thought. one time scripts (all tool scripts) suspend all other running global or model level scripts while running. are you running other scripts, lua mixers or lua widgets? can you please provide your radio and model yamls.

bastian2001, 2024-02-19:
Very good idea, I have the betaflight background LUA script. Removing that fixed this issue. Any idea why even with the background script, the D modes worked fine?

mha1, 2024-02-19:
No idea, you'd have to ask the BF guys.

schugabe, 2024-02-21:
That is no bug: we boost the TLM ratio if you are using lua scripts to make them more responsive independent of the tlm ratio you have selected.

mha1, 2024-02-21:
Just to understand: who is we? EdgeTX, ExpressLRS firmware, BF firmware, BF LUA script?

bastian2001, 2024-02-26:
For now, all I know is that the FC only receives half the amount of packets. @schugabe explained to me in the ExpressLRS Discord that if the Betaflight background LUA, the upstream telemetry is boosted for a few seconds, thus even at low transmission rates (only RTC sync between remote and FC) the RC packet rate is reduced. I haven't found the time to look into the data transmission between the module and the handset, whether the handset sends 1000 packets per second and the TX drops them or whether the handset itself already only sends 500 packets per second.

pkendall64, 2024-04-06:
Marked this for closing as it's not an ExpressLRS bug.

closed 2024-04-19 by pkendall64