ExpressLRS/ExpressLRS #2575: Telemetry lost at 50m after firmware update to 3.3.2 (T20 Gemini+RP4TD)

MR444, 2024-02-26:
Hi gys, yesterday I flashed my Jumper T20 Gemini and the RP4TD receiver with the newest firmware (3.3.2) with the configurator 1.6.1 via Betaflight pass-through.

I went flying on the same place like the day before but already had a lost telemetry after 50 meters and low dBm warning in the OSD before. The warning starts at -60dBm on my flight controller  I didn't fly further because I didn't want to failsafe.

The dBm at take-off is between -25 and -27 but after a few meters it dropps quickly to -70. Also indoors I have now the same problem. The day before I could fly the longer distances indoors and outdoors even at 10mW instead of 100mW without any dBm warnings or lost telemetry.

So I tried 100Hz Full instead of 250Hz (-108) but still had the same problem, no change in the range.

Could telemetry logs help you? Then I can do a flight and upload the logs with my radio. Wanted to show you my OSD but the Goggles V2 unfortunately does not record data from the FC.

Any ideas? 

- TX hardware: Jumper T20 Gemini
- RX hardware: Radiomaster RP4TD
- ExpressLRS version  3.3.2, I chose "_Jumper Gemini T-20 2.4GHz TX_" and  "_RadioMaster RP4-TD True Diversity 2.4GHz RX_"
- Packet Rate: 100Hz full and 250Hz (-108)
- Telemetry Ratio: 1:8
- Switch mode: Wide
- Antenna mode: Gemini
- Model Match: Off
- Tx Power: 100mW
- Dynamic: Off

MR444, 2024-03-02:
With the help of another forum,, I think I found out what is the problem. The problem is the incompatible telemetry baud rate of the receiver and transmitter.

My Gemini receiver came with a telemetry baud rate of F1000 but the T20 has a stock telemetry of only 400k. This combination causes telemetry lost warnings

gagarinlg, 2024-03-02:
> So I had to change the baud rate of the transmitter to 921k (on page 6 "hardware" of EdgeTX) to be compatible with F1000 from the receiver. The manufacturers should change the stock settings since this is really confusing when you don't know these details about ELRS and always get telemetry lost warnings, even at 500mW transmission power just a few meters away,.

That is something we should probably change within EdgeTX for most radios. This part of your issue is (sadly ;) ) not to be blamed on ExpressLRS

closed 2024-03-25 by pkendall64

teteles, 2024-03-27:
> With the help of another forum,, I think I found out what is the problem. The problem is the incompatible telemetry baud rate of the receiver and transmitter.
> 
> My Gemini receiver came with a telemetry baud rate of F1000 but the T20 has a stock telemetry rate of only 400k. This combination causes telemetry lost warnings.
> 
> So I had to change the baud rate of the transmitter to 921k (on page 6 "hardware" of EdgeTX) to be compatible with F1000 from the receiver. The manufacturers should change the stock settings since this is really confusing when you don't know these details about ELRS and always get telemetry lost warnings, even at 500mW transmission power just a few meters away,.

Thank you for your report, I'm having problems with my brand new T20 GEMINI unit, I took it out of the box and started the process to bind it but I was stuck in the "Loading" of the Lua Script. I tried all the tutorials I found, for example;

1_Set to the HARDWARE page> Internal RF:CRSF

2_Create new MODEL-01> Internal RF CRSF: 115k,400k,921k and 1.85M

3_I tried the factory lua script and the new one that I generated with the password pass.

unsuccessfully, keeps loading forever...

Could you please make the files that worked on your radio available for me to test here?

MR444, 2024-03-27:
When your LUA script is not loading, it looks like your radio has another problem. I just used the one that came with the radio. Can you show where you were stuck and what was the error?