ExpressLRS/ExpressLRS #2577: 3.3.2 - Link lost with Matek R24-P6 PWM receiver

0crap, 2024-03-03:
## Current Behavior

It happened twice I got a complete link loss with a [Matek R24-P6](https://www.mateksys.com/?portfolio=elrs-r24-p6) PWM receiver.

## Steps to Reproduce

This is the biggest issue, I have no clue how to reproduce.
Flying indoor (gym) with my foam airplane it happened once. (Doing 6 packs or something.)
Happened one more time during bench testing.

## Possible Solution (Not obligatory)

- 

## Details

Matek R24-P6 PWM receiver running 3.3.2
Only config change is 400Hz PWM for the ESC

When the link loss happens the receiver seems to do a complete reboot from what I can tell.
Failsafe values are not applied.
ESC had a little bit throttle when rolling over the gym floor towards me. Then the link loss happened and the plane kept rolling over the floor while I had no control at all.
Taranis gave the "Telemetry lost" message.
After 10-15 seconds the link came back automatically.

## Your Environment

- TX hardware: DIY E28 3.3.2 ISM2G4
- RX hardware: Matek R24-P6 PWM
- Handset model: Taranis QX7S (External RF @ 1.87MBps)
- OpenTX version: 2.3.15
- ExpressLRS version 3.3.2
- Packet Rate: 100Hz Full
- Telemetry Ratio: Std
- user_defines:

[image]
[image]
[image]

pkendall64, 2024-03-03:
This is a duplicate of #2548 
The problem is that the 8285 based receivers do not have hardware PWM output like the ESP32 based receivers, so the PWM is controlled using a timer and an interrupt. The reboot occurs because somehow the receiver is blocking in the timer interrupt and not returning causing a watchdog timer to fire.
I have worked around the issue in #2571 and run a 48 hour test on the bench without issues. See the PR for more details.

closed 2024-03-03 by pkendall64

0crap, 2024-03-03:
> This is a duplicate of #2548 The problem is that the 8285 based receivers do not have hardware PWM output like the ESP32 based receivers, so the PWM is controlled using a timer and an interrupt. The [...]

Please add this information as a warning or known issue on the [releases](https://github.com/ExpressLRS/ExpressLRS/releases) page, as it can be very dangerous.
Good to see the issue is worked at, thx.
"Workaround" sounds a bit scary, would be nice if the root cause can be found...