ExpressLRS/ExpressLRS #2584: Remapping option for channel 5 in SBUS

Kjblootens, 2024-03-08:
### Discussed in https://github.com/ExpressLRS/ExpressLRS/discussions/2479

<div type='discussions-op-text'>

<sup>Originally posted by **apiening** November 19, 2023</sup>
I have an airplane from **FMS** which has a **stabilizer** system (**Reflex V2**) which accepts **SBUS**. I enabled SBUS output for my RX (**ELRS 3.3**) and the control link works fine.
However, this stabilizer uses channel 5 with a three position mode switch to select between stabilizing modes (auto level, optimized, off). So when I configure it this way, I also disarm when I choose auto level stabilized mode.

I can assign the **PWM** outputs freely with **ELRS**, but is there a chance to **remap channel 5** for the **SBUS** output?</div>

stevewitwitty, 2024-03-30:
I use SBUS with a Helicopter with the same issue. Helicopter uses channel 5 for stabilization mode. I got around this issue by going into the Express LRS lua script and changing the packet rate to 333Hz and then changing the switch mode to 16ch Rate/2. This then allows full resolution on channel 5 and the helicopter's stabilization mode switches on and off via a switch as needed. You can read further info here: https://www.expresslrs.org/software/switch-config/?h=switch

stevewitwitty, 2024-03-30:
Forgot to say, there is a video which i found that helped me with this:

https://www.youtube.com/live/ARCfafma1rM?si=6MxiHAZJrRpC9g_N

Credit to RC Video Reviews for the video

CapnBry, 2024-05-23:
A better solution is to allow remapping of all channels on the receiver. Not just SBUS not just CH5. This would let the user decide not only where CH5 ends up coming out, but also how the other channels are shifted.

Kjblootens, 2024-05-24:
CapnBry. Yes that's implied here but still a good call out.

The problem only really exists for CH5 though. For any other issues there are very viable workarounds in terms of channel remapping on the TX.

The only viable workaround here are 16/2 Full modes where CH5 is not a prioritized channel along with CH1-4 as ALL packets get deprioritized (as per the switch config page, 16/2 modes have no lines bolded.) https://www.expresslrs.org/software/switch-config/

The ask is to enable serial output remapping (any serial protocol) so we can avail of the benefits of other switch modes while maintaining use of CH5 for e.g. flight mode switching where the target system (stabilizer) cannot be modified.

CapnBry, 2024-05-24:
Yeah that's what I was saying, remap all channels to any order in the webui like you can do for pwm outputs.

I don't think we'll ever "deprioritize CH5" because it is how thousands of people have their stuff set up currently, but remapping the channels means that it doesn't matter because you can have the output CH5 be whatever, and 16ch mode (which kinda sucks) won't be as used. I think the major reason people use 16 channels at half rate is just for CH5.

Gerold68, 2024-06-04:
CapnBry,
>I think the major reason people use 16 channels at half rate is just for CH5.

You underestimate the gilder guys that use more than 8 Channels! 16/2 is made for us ;-)

CapnBry, 2024-06-04:
> You underestimate the gilder guys that use more than 8 Channels! 16/2 is made for us ;-)

I hear ya! But CH5 is still going to be ARM no matter what, until it gets removed (in 2030 maybe?). Until then you'll still only get 15 channels unless you're not using of the things that rely on CH5, but you can do that currently without any ExpressLRS changes. Unfortunately, the proposed RX-side mapping won't work with any of the glider scripts that rely on the handset's channel order. :/

ParkJeff9, 2024-12-13:
My main use case is fixed wing. and I would like to add my support for channel re-mapping in SBUS and other serial outputs. I don't use the workaround because, I believe, not using ch 5 for arming messes up dynamic power in the Tx. Better if the rx can map another channel to the ch 5 position in the serial data stream.

mha1, 2024-12-13:
> My main use case is fixed wing. and I would like to add my support for channel re-mapping in SBUS and other serial outputs. I don't use the workaround because, I believe, not using ch 5 for arming [...]

V4 and EdgeTX 2.11 will solve the CH5 SBUS inconvenience. Read through https://github.com/ExpressLRS/ExpressLRS/pull/3008#issue-2613672057 for more info on the new arming mode. This will get you channel 5 as full proportional channel in all full-res modes and also for SBUS output while still being able to independently arm the module. No need for remapping or fiddling with ELRS code to remap SBUS.

To play with this use Configurator to build and flash branch master for your rx and tx module. And make sure to flash a recent EdgeTX nightly to your EdgeTX radio (e.g. https://github.com/EdgeTX/edgetx/releases/download/nightly/edgetx-firmware-nightly-0ea41f21.zip)

ParkJeff9, 2024-12-14:
> V4 and EdgeTX 2.11 will solve the CH5 SBUS inconvenience. Read through [#3008 (comment)](https://github.com/ExpressLRS/ExpressLRS/pull/3008#issue-2613672057) for more info on the new arming mode. This will get you channel 5 as full proportional channel in all full-res modes and also for SBUS output while still being able to independently arm the module. No need for remapping or fiddling with ELRS code to remap SBUS.

That would be excellent. Thank you.

airbass70, 2025-10-17:
> V4 and EdgeTX 2.11 will solve the CH5 SBUS inconvenience. Read through [#3008 (comment)](https://github.com/ExpressLRS/ExpressLRS/pull/3008#issue-2613672057) for more info on the new arming mode. This will get you channel 5 as full proportional channel in all full-res modes and also for SBUS output while still being able to independently arm the module. No need for remapping or fiddling with ELRS code to remap SBUS.
> 
> To play with this use Configurator to build and flash branch master for your rx and tx module. And make sure to flash a recent EdgeTX nightly to your EdgeTX radio (e.g. https://github.com/EdgeTX/edgetx/releases/download/nightly/edgetx-firmware-nightly-0ea41f21.zip)

When a new tag gets released, does it originate to the current branch master? Otherwise, can some commits be excluded? 
example: PR 3008 was merged with master branch in nov 2024 but it seems to be not included yet in the later tags (I cant find any of its commits in the 3.6). On the other hand, PR 3008 comes with a label V4.0 which let me think it is possible to merge it and keep it on hold. Can somebody explain how does it work?  Sorry, I know this is git dumb question. The reason being I am learning github and waiting for CH5 full res on an official release. Thank you developers for understanding.

mha1, 2025-10-17:
https://github.com/ExpressLRS/ExpressLRS/pull/3008 is supported by EdgeTX 2.11

r1000ru, 2026-01-20:
Is this decision reflected in the CRSF protocol? Or is it invented by the developers of ELRS and EdgeTX and will not be implemented by hardware manufacturers on other operating systems that support the official CRSF?

mha1, 2026-03-19:
> This doesn't seem to have been resolved with ELRS 4.0.0. I'm testing an SBUS output which seems to still be tied to the arm/disarm signal, (even after making the settings change in edgetx) as confirmed with a logic analyzer. Here's the TX configuration [image] And here's a recording in saelea logic demonstrating that channel 5 is still being toggled when I press the arm/disarm switch. [SBUS arm Switch toggle recording.sal.zip](https://github.com/user-attachments/files/26109607/SBUS.arm.Switch.toggle.recording.sal.zip)
> 
> In the edgetx outputs view, there's no indication of any changes to channel 5 when I arm or disarm, and switch configured input for channel 5 does show as changing the channel 5 output, though that seems to not really be the case.
> 
> Tested with Radiomaster TX15 Running Edgetx 2.12.0-RC4.

SBUS outputs the full proportional value of EdgeTX mixer for CH5 if:
- ELRS version is min 4.0.0
- EdgeTX version is min 2.11.0
- Arm using is set to Switch
- packet rate is set to 100Hz full res or 333Hz full res