ExpressLRS/ExpressLRS #2596: Unexpected ppm channel 9

qubicllj, 2024-03-14:
## Current Behavior

I have build a diy e28 tx module, it works on my futaba T10J, but i have noticed the unexpected channel 9 on betaflight receiver tab.

## Steps to Reproduce

1. Attach DIY E28 module to Futaba T10J
2. Power up Futaba T10J
3. Power up quad (happymodel ep2)
4. Check channel status in betaflight receiver tab

## Possible Solution (Not obligatory)

I have modified the source code of PPMHandset.cpp as below:

`for (int i = 0; i < length - 1; i++)`

The problem seems solved.

## Details

## Your Environment

- TX hardware: DIY E28
- RX hardware: Happymodel EP2
- Handset model: Futabal T10J
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): tx master 275f94ce58e547b9ffe08cfbf4f7d3b7924e55e8, rx V3.2.1 
- Packet Rate: 150Hz
- Telemetry Ratio: 1/32
- user_defines:

pkendall64, 2024-03-15:
Can you provide a trace of the signal on the PPM line. The code as it is is working correctly on the EdgeTX based radios.
I have tried by setting the number of channels in EdgeTX to 4 and 5 and the code detects the correct number of channels and when it's less than 5 it uses auto arm and if 5 or more it is manual arming via channel 5.

qubicllj, 2024-03-17:
I have add debug line as below:

`DBGLN("channel %u : duration0 %u, duration1 %u", i, (uint8_t)item.duration0, (uint8_t)item.duration1);`

The result:
channel 0 : duration0 161, duration1 177
channel 1 : duration0 160, duration1 177
channel 2 : duration0 161, duration1 187
channel 3 : duration0 161, duration1 202
channel 4 : duration0 161, duration1 67
channel 5 : duration0 161, duration1 68
channel 6 : duration0 160, duration1 68
channel 7 : duration0 161, duration1 67
channel 8 : duration0 161, duration1 0

pkendall64, 2024-03-17:
By casting to uint8_t you've lost the top bits of the data! I was wondering why these values looked so wrong.
Can you redo this with the cast removed. Also it would be good to get the level associated with each duration as well.
An oscilloscope/logic analyser trace would be perfect if you can get it.

pkendall64, 2024-03-19:
Thanks for the information. It looks like we should just ignore items that have a zero duration as you seem to be getting a trailing pulse 4000 ticks then some spike of 36 ticks.

pkendall64, 2024-03-26:
@qubicllj I have a PR in place to fix the issue properly. Could you test this on your Futaba radio for us please.

qubicllj, 2024-03-26:
Ok, i will test it in 10 hours.

PR #2612 (2024-03-28): Fix TX module PPM input seeing spurious channels
Ignore trailing items with 0 duration. This fixes spurious channel data for inverted PPM and glitches.
Fixes #2596
closed 2024-03-28 by JyeSmith