ExpressLRS/ExpressLRS #2635: PWM limits insufficient to drive servos to full limits of operating range

nick-cnc, 2024-04-04:
## Current Behavior

My ExpressLRS radio can not drive my servos to the full range of travel that my Sanwa or Flysky radios can achieve.

## Steps to Reproduce

1. Hook a servo up to a Flysky radio
2. Use the radio to drive the servo to its physical limits
3. Note the positions of those limits
4. Disconnect servo from Flysky radio and hook up to an ExpressLRS radio
5. Use the radio to drive the servo to the limits of ExpressLRS
6. Note the positions of those limits (they will be notably less than the limits observed in step 3)

## Possible Solution (Not obligatory)

Extend the PMW pulse length limits to at least 900us to 2100us

## Details

I am unable to set my rc car up with the full range of steering angle that I am able to achieve using my Flysky NB4 radio.

Servo in question: https://www.savoxusa.com/products/savsc1257tg-std-size-coreless-digital#technical-details

please note under the technical specifications it lists a maximum travel range of 900 to 2100us which, as far as I can tell, is beyond the limits set by ExpressLRS.

## Your Environment

- TX hardware: Radiomaster MT12 internal RF module
- RX hardware: Radiomaster ERC3-i
- Handset model: Radiomaster MT12
- OpenTX version (including nightly number): EdgeTX 2.10 - can't remember off the top of my head which nightly it is
- ExpressLRS version (TX & RX MUST MATCH): either 3.3.1 or 3.3.0, can't remember which off the top of my head
- Packet Rate: F1000
- Telemetry Ratio: no clue
- user_defines: again, no clue

mawzthefinn, 2024-04-05:
> Just enable EdgeTX "Extended limits" and set channel min/max to -120/120. https://edgetx.gitbook.io/edgetx-user-manual/edgetx-user-manual/user-manual-for-color-screen-radios/model-settings/inputs-mixes-and-outputs/outputs ![image](https://private-user-images.githubusercontent.com/39406583/320094527-d7770e91-516f-43d2-a11f-8bf37910623c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzNDAyMDcsIm5iZiI6MTcxMjMzOTkwNywicGF0aCI6Ii8zOTQwNjU4My8zMjAwOTQ1MjctZDc3NzBlOTEtNTE2Zi00M2QyLWExMWYtOGJmMzc5MTA2MjNjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDE3NTgyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJkZGFiOTEyMzA3ZjNlMmJmNGFjMzE2NjVkYzg0ZWFlZTcwOTg2NDgwNWRjOTUzMWQ2NzgzNGFiNWI0ZGRlNzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.v_nQ20QSxDauKoowSln94ksPb5v7e6UbO5aAZefmxv8)

That won't work, as the default limits are already sufficient to get to <900uS, you need extended limits when you want to go beyond the 885->2115uS range, not when you want to exceed 988->2012uS.

mawzthefinn, 2024-04-05:
> ![image](https://private-user-images.githubusercontent.com/39406583/320106316-192ea9da-246f-4874-bf87-502de916ad15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzNDMzMzQsIm5iZiI6MTcxMjM0MzAzNCwicGF0aCI6Ii8zOTQwNjU4My8zMjAxMDYzMTYtMTkyZWE5ZGEtMjQ2Zi00ODc0LWJmODctNTAyZGU5MTZhZDE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDE4NTAzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU1YzZhZTA5MjM3N2U1YzEzMWE2ZTc0NzBlYjFkZDkxMTA4NjBiYTE5OGE2ZTA4NDliNmZmZWQ3ZGZmNDc1NWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AWqwOYoPh5iQbo0APTo681Z651reb8PHp09xWpENGQw)
> 
> https://www.expresslrs.org/software/switch-config/

ANd that's the actual fix. Not extended limits (which does nothing in F1000 mode)

nick-cnc, 2024-04-05:
> Just enable EdgeTX "Extended limits" and set channel min/max to -120/120. [...]

I do have extended limits enabled. The servo stops responding somewhere around 105% in both directions. EdgeTX shows it is sending out pulse lengths as low as roughly 750us and as high as roughly 2200us, but the servo stops moving at around 980us on the low end and 2000us on the high end.

nick-cnc, 2024-04-05:
> ![image](https://private-user-images.githubusercontent.com/39406583/320106316-192ea9da-246f-4874-bf87-502de916ad15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzNDQ4ODgsIm5iZiI6MTcxMjM0NDU4OCwicGF0aCI6Ii8zOTQwNjU4My8zMjAxMDYzMTYtMTkyZWE5ZGEtMjQ2Zi00ODc0LWJmODctNTAyZGU5MTZhZDE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDE5MTYyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2YmZiMjViOWUxMGYwYWRlNjFjYTY0MzIwN2RmYWRkZGVjOTAwNWQ5YTA5YTk3YmIyOWFkZTAxYWQxZWI1YWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.G52Y1LQcq0AVT4gX27XfvkDY1wgAcd8MZaSdKewRx1E)
> 
> https://www.expresslrs.org/software/switch-config/

Not trying to sound combative, but this is way too technical and dense for a layperson like me to understand. Can you please explain what you're trying to tell me in a way that someone who ISN'T an ELRS dev could understand?

mawzthefinn, 2024-04-05:
> Not trying to sound combative, but this is way too technical and dense for a layperson like me to understand. Can you please explain what you're trying to tell me in a way that someone who ISN'T an [...]

Drop to 333Hz if you want to get the wider PWM limits.

nick-cnc, 2024-04-05:
> > > ![image](https://private-user-images.githubusercontent.com/39406583/320106316-192ea9da-246f-4874-bf87-502de916ad15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzNDQ4ODgsIm5iZiI6MTcxMjM0NDU4OCwicGF0aCI6Ii8zOTQwNjU4My8zMjAxMDYzMTYtMTkyZWE5ZGEtMjQ2Zi00ODc0LWJmODctNTAyZGU5MTZhZDE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDE5MTYyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2YmZiMjViOWUxMGYwYWRlNjFjYTY0MzIwN2RmYWRkZGVjOTAwNWQ5YTA5YTk3YmIyOWFkZTAxYWQxZWI1YWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.G52Y1LQcq0AVT4gX27XfvkDY1wgAcd8MZaSdKewRx1E)
> > > https://www.expresslrs.org/software/switch-config/
> > 
> > 
> > Not trying to sound combative, but this is way too technical and dense for a layperson like me to understand. Can you please explain what you're trying to tell me in a way that someone who ISN'T an ELRS dev could understand?
> 
> Drop from F1000 to 333Hz and set switch mode to 8ch if you want to get the wider PWM limits.

This seems like a bandaid and not a real solution. Is there a technical reason why extended limits can't be used in F1000 mode?

mawzthefinn, 2024-04-05:
> This seems like a bandaid and not a real solution. Is there a technical reason why extended limits can't be used in F1000 mode?

CRSF Extended limits are only supported in 8ch, 12ch Mixed and 16ch/2 modes

Those modes only support 50Hz, 100Hz and 333Hz packet rates. 

It's all documented in the link above.

nick-cnc, 2024-04-05:
> CRSF Extended limits are only supported in 8ch, 12ch Mixed and 16ch/2 modes
>
> Those modes only support 50Hz, 100Hz and 333Hz packet rates.
>
> It's all documented in the link above.

So are you saying it is not possible to extend the PWM range in F1000, or that it simply your opinion that it should not be done?

mawzthefinn, 2024-04-05:
> > > > > > https://www.expresslrs.org/software/switch-config/

> 
> So are you saying it is not possible to extend the PWM range in F1000, or that it is simply your opinion that it should not be done?

If you read the docs at that link, it will tell you which output range options are supported in which switch mode, and which switch modes are supported in which packet rates.

pkendall64, 2024-04-05:
CRSF protocol sends an 11 bit value from the handset to the ELRS module, where 10 bits is 988-2011. -100% to 100% range.

The hybrid and wide switch modes in ELRS have 10bits of channel data. Which directly translates the -100% to 100% range over the air. 2011-988 = 1024 (values 0-1023) which is 10bits.

8/12/16 channel modes also use 10bit over-the-air but divide the 11bit raw value by 2 and sends, then extends it again on the receiver, by multiplying by 2. So you can get the full range of values.

nick-cnc, 2024-04-06:
Right, I feel like I'm not being heard.

Extended PWM Limits in F1000 mode: impossible or not? will be done or not?

I understand that's not how ELRS currently operates, and I understand there may be reasons it wasn't done that way before. But I'm asking now - can it be done, and if so, will it be done? That's all I want an answer to.

pkendall64, 2024-04-06:
It's not possible with the current F1000 RF mode and the current switch modes.

It would be possible with a new switch mode or with a different RF mode with parameters that would allow the 8/12/16 switch modes.

Can it be done? Yes.
Can it be done with the current F1000 parameters? No.
Will it be done? Who can tell, just needs someone to do it.

closed 2024-04-19 by github-actions[bot]