ExpressLRS/ExpressLRS #2639: subs output become elrs output

yangmoulalala, 2024-04-05:
## Current Behavior

Hello, my receiver device is the BETAFPV 2.4GHz Nano RX, Firmware Rev. 3.3.2 (2d9891) ISM2G4. It used the SUBS output feature, but for some reason, after a reboot during usage, the receiver switched to ELRS output. This has happened twice. I did not manually change its settings.
## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2024-04-05:
Only way tat could happen is if the receiver thinks the boot button is being help down and is resetting the configuration.

pkendall64, 2024-04-07:
I've tested this by flashing a BETAFPV 2.4GHz Nano RX with 3.3.2 and power cycled the receiver a good dozen times and it has never reverted back to CRSF from SBUS mode.

Can you provide any more information about your setup?

yangmoulalala, 2024-04-07:
Hello, thank you very much for your reply. After several attempts, I've found that this issue usually occurs when the drone suddenly loses control and flips over during flight. I now suspect it might be my mistake. I'm using the PX4 firmware, and I've mounted the receiver and the drone's telemetry module in the same location, which might have caused some interference between them. Next, I plan to move the receiver away from the telemetry module and try again. If the issue persists, I will contact you again. Thank you for your openness.

yangmoulalala, 2024-04-07:
> Only way that could happen is if the receiver thinks the boot button is being help down and is resetting the configuration.

Sorry for only seeing your message now. So that's what happened. I'll go check my receiver installation.

pkendall64, 2024-04-07:
Thanks for getting back to me. I'll mark the ticket as closing, which will give us 14 days to see if the issue can be reproduced.

closed 2024-04-20 by github-actions[bot]