ExpressLRS/ExpressLRS #2644: Missing barometer 'Alt' value in telemetry data on the radio ...

nostromo123, 2024-04-08:
According to 
https://github.com/ExpressLRS/ExpressLRS/tree/master/src/lib/CrsfProtocol 
I assumed, I should see barometer 'Alt' on the radio telemetry screen.

- TX hardware: FrSky X10 S Express with HiYounger ELRS modul
- RX hardware: BETAFPV 2.4GHz Light RX
- OpenTX version: EdgeTX 2.9.4
- ExpressLRS version (TX & RX MUST MATCH): 3.3.2 TX and RX
- Packet Rate: 333Hz Full
- Telemetry Ratio: 1:4
- user_defines: FC SpeedyBee F405 Wing mini with INAV 7.1.0

I get those telemetry values here on my radio

[image]

Changing over to FrSky TX/RX communication with RX Archer RS in F.Port mode, I do get 'VSpd' as well as 'Alt' from the FC barometer

[image]

br KH

mha1, 2024-04-09:
not an ExpressLRS issue.

closed 2024-04-09 by mha1

Target0815, 2024-04-09:
According to this [issue](https://github.com/iNavFlight/inav/pull/9438), INAV have implemeted ID9 for CRSF in version 7.1 **RC1**. Then it should also work ...

mha1, 2024-04-09:
and this https://github.com/iNavFlight/inav/pull/9858 seems to have reverted it back to ID7 two weeks ago

nostromo123, 2024-04-10:
and with FrSky transmission the 'Alt' value shows up, but not with ELRS!

mha1, 2024-04-10:
well, different protocols.

Again, this is not an ELRS problem. What's not sent can't be displayed, agreed? iNav uses CRSF packet ID7 to send baro data. CRSF Packet ID7 doesn't contain Alt. So iNav is not sending Alt. 

iNav obviously tried it with ID9 but as they state some radios weren't prepared to receive ID9 (which carries Alt). That's why they went back to ID7 one week prior to the 7.1.0 release. Just read the links above.

All I can say is EdgeTX is prepared, don't know about others and when they are going to update their radio's firmware. I'm sure iNav will switch to use ID9 as soon as the non-preppers they have in mind have updated their firmware.

nostromo123, 2024-04-10:
And all can say

INAV + FrSky OK and INAY + ELRS NOT OK.

I don't think, it is my task to present a solution, when posting here but I might be wrong.
Forget it and have a nice day .............

mha1, 2024-04-11:
The solution is simple and it can only be implemented by the iNav project. iNav needs to send ID9 packets instead of ID7 packets. This is why I asked you to raise an issue with iNav. ELRS can't solve this.

nostromo123, 2024-04-11:
Excuse my persistence Michael. You did not say **anything!** to that screenshots here. Left ELRS and on the right FrSky HF transmission between the **same** radio and **same** INAV on the **same** SpeedyBee.

[image]

But thank you for not pointing me into the FrSky direction, because FrSky shows the 'Alt' value, which in opinion is not supposed to be there.

But as you closed it anyway .....

mha1, 2024-04-11:
I am sorry, I'm obviously not capable to explain the problem for you to follow.

Please accept the fact that FrSky and CRSF use two completely different protocols to communicate telemetry data. iNav provides Alt to FrSky receivers using FrSky protocol, but iNav doesn't provide Alt to ELRS receivers using CRSF protocol protocol. ELRS receivers can't be set to use FrSky protocol, they require CRSF protocol.

Again, please raise an issue with iNav. iNav needs to send Alt using the CRSF ID9 packet (they'll understand what that means) in order for you to see Alt using an ELRS receiver.

pkendall64, 2024-04-11:
The other issue that compounds this is that Frsky Ethos radios do not understand CRSF ID9 only ID7. Probably why iNav reverted the PR.
There was a similar issue 2349 raised and closed with PWM receivers with a barometer. ELRS outputs CRSF ID9 because we have altitude and vertical speed, Ethos only reads ID7 on CRSF protocol. i.e. Ethos issue.

Merga189, 2024-04-12:
Sorry for me, not understanding at once, what is the issue here. But pointing to SpeedyBee at the beginning here ...

I'll post a link to this closed issue discussion at  https://github.com/ArduPilot/ardupilot/issues/26699 and hope, someone there reads and understands.

Not much hope.