ExpressLRS/ExpressLRS #2667: RP4TD not appearing in lua script

JGrundon, 2024-04-19:
## Current Behavior
With Radiomaster RP4TD bound to Radiomaster Ranger Micro there is no "other devices" folder in the lua and therefore no way to change Rx settings from the lua.

## Steps to Reproduce
Bind RP4TD to Radiomaster Ranger Micro

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details
Rx binds to Tx and "C" can be seen in the top right of the lua. Rx seems to work fine other than the described issue. I have also tried another RP4TD and get the same issue. I have multiple RP1s and they show up in the lua with no issues.

## Your Environment

- TX hardware: Ranger micro
- RX hardware: RP4TD
- Handset model:TX16S
- OpenTX version (including nightly number) EdgeTX 2.9.0
- ExpressLRS version (TX & RX MUST MATCH): 3.3.2
- Packet Rate: 100Hz
- Telemetry Ratio: 1:32
- user_defines:

JyeSmith, 2024-04-21:
Can you please test the 3.4 RC2 and report if it is still happening.

pkendall64, 2024-04-21:
I have seen this sometimes if you enter the Lua script before the RX has actually connected.
If you exit and re-enter the Lua script it should work. I have verified that it does show on the TX16S with RP4-TD.

JGrundon, 2024-04-22:
> I have seen this sometimes if you enter the Lua script before the RX has actually connected. If you exit and re-enter the Lua script it should work. I have verified that it does show on the TX16S with RP4-TD.

I tried letting everything sit for few minutes after connecting before entering the lua but there was still no "other devices" folder.

JGrundon, 2024-04-22:
> Can you please test the 3.4 RC2 and report if it is still happening.

I updated just the Tx to 3.4 RC2 and this fixed the problem. Yay now I can turn on diversity! Any idea when the final version will release? Thanks so much for your help.

pkendall64, 2024-04-22:
It will release when all the known bugs are resolved and PR's merged and testing does not reveal any show stoppers.
So real-soon-now™

closed 2024-04-22 by pkendall64