ExpressLRS/ExpressLRS #2669: Screen flashes unexpectedly when running elrsV3 Lua script

RC-SOAR, 2024-04-20:
## Current Behavior

When running elrsV3 script on my TX16S under OpenTX 2.3.15, the screen flashes rapidly after each scroll increment, which is very distracting.  Video here:

https://youtu.be/WO_e5PuhHwU?si=PkAvhA9-y2YpKBXa

The actual link with the rx works fine.

Is there a Lua fix for this?

(I notice that the "configure Crossfire" script in the Tools menu provides similar functionality and doesn't suffer from the flashing issue.)

## Steps to Reproduce

1. Go to OpenTX Tools menu
2. Run ELRSv3 script

## Your Environment

RM TX16S
OpenTX 2.3.15
Ranger Micro external ELRS 3.3.0

pkendall64, 2024-04-21:
This looks like an openTX bug as this does not happen on EdgeTX with the same Lua script.

RC-SOAR, 2024-04-21:
Further data point... version 3.2 works fine (after selecting '3.2' release in the configurator and downloading the script). So an incompatibility has crept in with 3.3 script. 
It would be great if this were investigated as there are still many OTX users out there.

aerostarlegacy, 2024-04-24:
Hello! This is effecting me as well.

JyeSmith, 2024-04-24:
OTX is no longer maintained or supported.  The fix here is to update to EdgeTX.

RC-SOAR, 2024-04-24:
Confused, as ELRS apparently supports OpenTX 2.2.12 or newer as in:
https://www.expresslrs.org/quick-start/transmitters/tx-prep/#internal-expresslrs-modules
The closing of this issue appears to contradict this.

In the meantime, I worked around this by using the 3.2 script (as advised on Discord). Seems to work, but I have no idea what's changed.

pkendall64, 2024-04-25:
The changes from the 3.2 script are basically to add support for Ardupilot and some optimisations to reduce memory usage for memory limited radios, some fixes to better support configuration of PWM receivers and reducing screen update so it loads faster (this was because of change in the EdgeTX graphical library).
So if the older version works for you then just keep using it.

RC-SOAR, 2024-04-25:
Thanks Paul. Given that there are still lots of OTX users out there it would be great if compatibility were maintained if poss, though I appreciate it's an increased maintenance burden for you guys.

aerostarlegacy, 2024-04-25:
@RC-SOAR - can you post a link to that 3.2 script here so others might find it?

RC-SOAR, 2024-04-25:
@aerostarlegacy I downloaded it via the Configurator, after selecting the 3.2 target.

CapnBry, 2024-05-23:
> Given that there are still lots of OTX users out there it would be great if compatibility were maintained if poss

Why are they still on OTX? EdgeTX is available for all OTX targets. OpenTX is dead and EdgeTX has replaced it.

RC-SOAR, 2024-05-23:
OTX on a TX16S is **much** more productive when dealing with setups with 60 or so mixers as is not uncommon with sailplane setups like my [F5J template](https://rc-soar.com/edgetx/setups/esoarplus/index.php)

mha1, 2024-05-24:
What does you mean by productive?

RC-SOAR, 2024-05-24:
> What do you mean by productive?

The menus in OTX are denser, so less scrolling and swiping, and the mixers menu caters well for complex setups.
The Mixers menu in ETX is especially awkward to manage with large numbers of mixers - the UI is slow and tedious. It's as if ETX was designed primarily for use with FCs. 
All this applies to the colour screen radios only, the non-touch mono screen interface is fine.

closed 2024-06-19 by github-actions[bot]

macgyverek, 2024-08-31:
Added fix on my repo for that issue: https://github.com/macgyverek/ExpressLRS-LUA-OpenTx

RC-SOAR, 2024-08-31:
Thanks for this. Out of interest, can you point to the corrected lines of code?

macgyverek, 2024-08-31:
Yes, no problem: 
[image]

RC-SOAR, 2024-08-31:
It would be nice if the fix could be applied to the production version. Any chance?

CapnBry, 2024-09-01:
> It would be nice if the fix could be applied to the production version. Any chance?

Nope, because making that change absolutely _crushes_ performance on EdgeTX. It redraws the whole screen every update which is crazy slow on EdgeTX to the point it takes 2-3x longer for the list to load.

macgyverek, 2024-09-01:
But you can use version checker and write if/else for edgetx, opentx. But I know...developers are very stubborns ;)

CapnBry, 2024-09-03:
> But you can use version checker and write if/else for edgetx, opentx. But I know...developers are very stubborns ;)

There's incredibly low RAM available on many platforms the lua runs on. Having the script try to detect and run two different ways is a waste of resources that it doesn't have. 

Feels more like someone could just update their supported hardware to a version of EdgeTX and resolve this themselves, but users are very stubborns about solving their own problem ;)  And why doesn't the Configurator support Windows Me?! 😆

macgyverek, 2024-09-03:
EdgeTx it's crappy for user with colour screen without touch screen...like x10s, x12s... It's a very good radio, not "Chinese" Jumper...

More information are showed on screen with OpenTX. 
It's not update decision....if edgetx will make a screen (GUI) like in opentx, I will update today.