ExpressLRS/ExpressLRS #2710: [V3.4.0] Fails to build for BetaFPV Micro TX 900Mhz Version with EU_868_LBT

menschel, 2024-05-19:
## Current Behavior

When using Configurator in Version 1.6.1 and Selecting BetaFPV 900 --> Micro TX -> EU_868_LBT, apparently no firmware is build.

## Steps to Reproduce

1. Connect BetaFPV Micro TX 900Mhz Module to ELRS Configurator 1.6.1

2. Select BetaFPV 900, Micro TX, Legislation EU_868_LBT

3. Select Serial Port

4. Start Update

## Possible Solution (Not obligatory)

Wrong Target in Cloud Build.

## Details

ELRS Configurator falsely displays Error Message indicating Issue during Flashing which is not the case.
Issue happens when Binary is retrieved from Cloud.

I know that everything on my setup works. I updated the internal 2G4 module once to LBT and then to FCC.
Castrating the 250mW transceiver to 100mW by LBT firmware is a NO-GO.
I sincerely ask you to stop with this anticipatory obedience that has become common on Github ever since Microsoft took over regime. That is what made Nazi Germany possible in the first place!

## Your Environment

- TX hardware: BetaFPV Micro TX 900Mhz 500mW
- Handset model: Radiomaster TX12Mk2
- EdgeTX Providence 2.9.4
- ExpressLRS version (TX & RX MUST MATCH): 3.4.0

pkendall64, 2024-05-19:
It would be helpful if you could provide some details about the error message that is shown in configurator.
Perhaps discord in the #help-and-support channel may be of more help to you as well.

The 100mW limit for LBT is a requirement for the manufacturers to obtain a compliance certificate. It is up to you if you wish to flash LBT or FCC.

> I sincerely ask you to stop with this anticipatory obedience that has become common on Github ever since Microsoft took over regime. That is what made Nazi Germany possible in the first place!

Inflammatory comments, like the one above, are not very helpful or productive.

menschel, 2024-05-20:
Thank you very much.
The perspective to end up with a 500mW transceiver castrated to 100mW after updating to V3.4.0 is not feasible.
There was also the possibility to make the 100mW limit an option instead of enforcing it.

closed 2024-05-20 by menschel