ExpressLRS/ExpressLRS #2711: PWM value goes to 988 when powered on without TX connected

NorgeSquad, 2024-05-19:
The ER3C-I 3ch PWM Receiver 2.4GHz goes to 988 PWM when turned on. I assume thats a hardware issue. but is that something you guys can look into?

gagarinlg, 2024-05-19:
You have to set the failsafe values via the lua script.

NorgeSquad, 2024-05-19:
> You have to set the failsafe values via the lua script.

yes. Its all set to 1500.

pkendall64, 2024-05-19:
Information about version would be helpful!

NorgeSquad, 2024-05-19:
I just flashed it to 3.4.0 ELRS

CapnBry, 2024-05-22:
Are you sure it goes to 988, or are you saying "It moves all the way to one side and stays there until something connects"? The difference being that one is an uncommanded move, the other would be a bug where we are outputting 988us for some reason when we shouldn't. Can you check it with a PWM checker to verify?

NorgeSquad, 2024-05-23:
It moves all the way to one side and stays there until something connects. That's the right term yes :) I don't have any fancy equipment sadly

CapnBry, 2024-05-23:
Thanks for the extra information. This is likely due to the receiver hardware which puts out a bit of noise on the line before it boots to our code. Some servos take this is a valid pulse and move before we start driving the signal. Unfortunately there's nothing we can do about this if that's the cause.

CapnBry, 2024-05-28:
> Can you recommend a Rx that don't have this issue? I assume the ER5C-I has the same hardware?

Yeah anything that is 5 channels is going to have the issue (and also the Matek R24-P6), and pretty much everything that doesn't have a dedicated CRSF plug on it (because the bad stuff happens on those pins). So like the BetaFPV SuperP, or the Radiomaster ER6 or ER8 (not the ER6-G or ER6-GV as those use those pins for two of their outputs), ER8-G, ER8-GV. I think that might be it?

NorgeSquad, 2024-05-28:
Thank you. Will look into it :)

closed 2024-07-30 by github-actions[bot]