ExpressLRS/ExpressLRS #2715: after reboot rp1 reciver went into bind mode

petterm77, 2024-05-20:
## Current Behavior

after flight i changed battery and the reciver rp1 went into bind mode by it self.
had to go into the reciver via wifi hot spot and put in the bind phrase again to get it to connect to the transmitter

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

 

## Your Environment

 out in the field flying

- TX hardware:  radiomaster ranger tx
- RX hardware: radiomaster rp1
- Handset model: radiomaster tx-16s mkii
- OpenTX version edgetx 6.10 rc3
- ExpressLRS version (TX & RX MUST MATCH): 3.4 on both
- Packet Rate: 250 hz
- Telemetry Ratio:  std
- user_defines:

SunjunKim, 2024-05-23:
Can you try update v3.4.1?

pkendall64, 2024-06-06:
I think this is related to #2751, could you test the PR #2755 and report back in the PR please.

closed 2024-07-30 by github-actions[bot]