ExpressLRS/ExpressLRS #2723: VTX admin channel/band/send on switches

genapauk, 2024-05-22:
### Discussed in https://github.com/ExpressLRS/ExpressLRS/discussions/2712

<div type='discussions-op-text'>

<sup>Originally posted by **genapauk** May 19, 2024</sup>
Is there any possibility to bind VTX admin features (channel/band change and send new values to vrx and vtx) to my radiomaster switches, just like HT enable and HT start channel?

For example it's too long for me to push "sys" button, then find TBS agent lite or ExpressLRS lua script, then wait for options to render. It would be great if i could just bind these features to my switches, so i can switch between channels and bands with 1 and 2 button and send these values with button 3 (screenshots below).

It's not a feature request or idea future versions (but it would be great!!), i just wondering whether it's possible to do directly in code, because for now i have no idea where to start, would appreciate to get some clues or hints how and where i can do that.
I've already made some sort of my "custom" firmware versions of elrs with my own frequency domains in FHSS.cpp and unlocked sending vtx/vrx values with backpack even when drone is armed, so this feature would be a great addition for my work, thanks
[image]
Note! Its has to be exactly vtx admin features, i know switching channels can be done with switches even using betaflight, but i need to do it using my backpack to switch both vrx and vtx simultaneously!</div>

CapnBry, 2024-05-23:
I'm working on something similar to this, so it is coming.

MayanHawk, 2024-05-30:
This is possible using a combination of EdgeTX and Betaflight: https://www.youtube.com/watch?v=DkKr0d26C3A

I believe if the VTX is connected properly with MSP_VTX or SmartAudio or TrampTelemetry the video should apply

jiri-jirus, 2024-08-08:
betaflight ignores requests to change channel while armed

Akceptor, 2024-08-09:
> betaflight ignores requests to change channel while armed
This is a safety thing. But pretty easy to "fix" - it\s just one `if` statement in the code

Akceptor, 2024-08-22:
Is there at lest some API I can hit from Lua scripts to sed the desired values?

Akceptor, 2024-10-25:
I made a custom FW and it works https://www.youtube.com/watch?v=KGMa7ClkdXw
But I had to override some stuff in the existing TX (head tracker thing) because didn't want to go crazy with changes, and also I'm not a C++ programmer

aszczu, 2024-11-06:
> I made a custom FW and it works https://www.youtube.com/watch?v=KGMa7ClkdXw But I had to override some stuff in the existing TX (head tracker thing) because didn't want to go crazy with changes, and also I'm not a C++ programmer

Can you share your firmware? I'm looking for the same thing, but link under youtube video is 404.

pkendall64, 2026-09-06:
Closing as this is more of a feature request and it's been done in Lua, see https://github.com/ExpressLRS/Lua/tree/unified-lua-lsp#vtx-administrator-widget

closed 2026-09-06 by pkendall64