ExpressLRS/ExpressLRS #2737: X Lite Master slave Problem. With Same Bluetooth name

druckgott, 2024-05-26:
## Current Behavior

Today we want to use ttwo x Lite Pro both with edge tx 2.10.

As long as we both have the default Bluetooth Name we do Not get a connection. Then we changed the Name of the internal bluetooth Name we get a connection. 

## Steps to Reproduce

2 taranis x Lite with default Bluetooth Name and want to connect in Trainer Mode master and one slave. If you rename it you get a connection

## Possible Solution (Not obligatory)

Default Name contiains unique ID in it so the Name is Not exactly the Same name

## Details
Tales long time until we get the failute and can connect two Radios. 

## Your Environment

- TX hardware:x Lite Pro with internal Modul 2x
- RX hardware: Not relevant

mawzthefinn, 2024-05-29:
This sounds like an EdgeTX or PARA issue not an ELRS issue.

druckgott, 2024-05-29:
Sorry you are right 🤓

closed 2024-05-29 by druckgott