ExpressLRS/ExpressLRS #2746: Team Racing - Position and Aux assignment getting lost

MayanHawk, 2024-05-30:
## Current Behavior

Using the Team Racing mode, I have so far set and "lost" models on the bench. 

I have them setup to use AUX 5 (my 6 pos selector) and the proper position set to the quad. The first time I tested, everything worked. I tested a second time and two of the quads reset to Channel 7/disabled (default, non-configured settings). Subsequent attempts are as follows:

- Q1 only works if it is powered on first (flywoo)
- Q2 and Q3 behave like each other regardless of which one is powered first (Happymodel EP2)
- Q4 (Foxeer) works if it is powered on before Q2.
- If all four are powered on, one after another, Q3 will stomp on the other three
- Q1 and Q4 are the only ones garuenteed to work with each other nearly 100% of the time now
- If Q2 or Q3 power on (any order), Q1 and Q4 will become unreliable

Trying get any combination greater than three quads working at the same time is proving exceptionally difficult.

### **There is also no way to wipe the settings for team racing mode without flashing down to 3.3.2 then back to 3.4.1**

When I went to disable team race mode I started with Q1, moved to Q2 and the settings from Q1 were already applied. The same was the case for Q2. Q4 needed to be adjusted with the LUA. During the "turning off" process only one quad was powered at a time, and I exited the LUA between each quad as well.

## Your Environment

- TX hardware: Happymodel ES24 pro
- RX hardware: Foxeer Lite, Flywoo EL24E, Happymodel EP2
- Handset model: TX16s
- EdgeTX 2.9.0
- ExpressLRS version (TX & RX MUST MATCH): 3.4.1
- Packet Rate: D500
- Telemetry Ratio: STD
- user_defines: none

pkendall64, 2024-06-07:
It's possible that the PR may help with this too, if the config is getting reset completely.
If so, please let me know.

MayanHawk, 2024-06-07:
Config is not getting reset completely, just the assigned model position is being forgotten/lost

pkendall64, 2024-06-07:
I think it would still be worth a try to see if it solves the issue

closed 2024-07-30 by github-actions[bot]