ExpressLRS/ExpressLRS #2752: Can't use 3.4.x with RadioMaster Ranger Micro 2.4GHz

BRSS73, 2024-06-01:
## Current Behavior

After flashing 3.4.x firmware from 3.3.2 on my RadioMaster Ranger Micro 2.4GHz, it doesn't connect with RX and the latest LUA script stuck in "Loading...".
If I flash back with 3.3.x, TX and RX connect and the same LUA script work flawlessly.

## Steps to Reproduce

1. Flash RadioMaster Ranger Micro with 3.4.x

2. Update LUA script to latest version

3. Select a model that uses ELRS module

4. Launch LUA script

5. Die waiting for something to change on screen or RX to connect

## Details

I've tried to:
	- flash every single version of 3.4 (RC included) with ELRS configurator (either UART or WiFi)
	- flash with erasing before or not 
	- flush the firmware directory cache
	- use another computer
	- flash repartitioner.bin before flashing 3.4.x
	
I notice this line in the log when flashing v3.4.x: Warning: some reserved header fields have non-zero values. This image may be from a newer esptool.py?
There's not such warning with previous versions.

## Your Environment

- TX hardware: RadioMaster Ranger Micro 2.4GHz
- RX hardware: Not really revelant, but it's RadioMaster RP1 2.4GHz or RadioMaster RP2 2.4GHz
- Handset model: FrSky Horus X12S
- EdgeTX version: 2.10.1

psfales, 2024-06-12:
What baud rate (handset hardware settings) and packet rate (Lua script) are you using?
(It's not exactly the same signature, but it sounds similar the issue I had when going from 3.0.x to 3.2.x.    See https://github.com/ExpressLRS/ExpressLRS/issues/2010)   You may need to try increasing the baud rate or lowering the packet rate

BRSS73, 2024-06-12:
I'm using 115K and 150Hz.
I can't increase the baud rate as my X12S lacks the mod to achieve that (https://blog.seidel-philipp.de/fixed-inverter-mod-for-tbs-crossfire-and-frsky-qx7) and the packet rate is quite low.
I've had the tiny board in my workshop since, at least, one year. I think it's time for me to grab iron, solder paste, thin wires and install it.
Thanks for your help

psfales, 2024-06-13:
I would try lowering the packet rate, at least as an experiment to show that this is really the issue.   If that helps, then increasing the baud rate (with the inverter mod) would likely help too.

Are you sure that the X12S needs the inverter mod?

BRSS73, 2024-06-14:
Pretty sure. My X12S hangs within 3 seconds of changing the baud rate @400K.

Anyway, it doesn't hurt to try to lower the packet rate. I'll take your advice and do it tonight.

BRSS73, 2024-06-15:
Lowering packet rate wasn't enough.
I did my long-postponed homework and finally installed the mod.
I can now go @ 400K. Both TX and LUA script are fully functional.

Thanks @psfales for pointing me in the right direction.

closed 2024-06-15 by BRSS73

psfales, 2024-06-17:
I can't remember exactly which release this was, but a few versions later I think it stopped working for me at 400K, and I had to bump it up to the next step (921k?).   It may depend on the packet rate.