ExpressLRS/ExpressLRS #2757: BR1/BR3 WiFi not work on 3.4.2

zyren, 2024-06-03:
- TX hardware: Radiomaster Bandit/Micro/Nano
- RX hardware: BR1 & BR3
- Handset model: TX16S/Boxer/Zorro
- EdgeTX version: 2.10.1
- ExpressLRS version (TX & RX MUST MATCH): 3.4.2
- Packet Rate: 200Hz
- Telemetry Ratio: 1:32
- user_defines: -DRegulatory_Domain_FCC_915

We have tested all transmitters and receivers with firmware version 3.4.2, and the conclusions are as follows:

All 2.4GHz TX: WiFi works normally.

All 2.4GHz RX: WiFi works normally.

All 900MHz TX: WiFi works normally.

All 900MHz RX (BR1/BR3): WiFi does not work. Despite attempting to activate WiFi in various ways (via Lua, holding a button for 5 seconds, waiting for 60 seconds), the receiver's light flashes rapidly indicating WiFi mode is correctly activated, but the SSID: ExpressLRS RX cannot be found.

pkendall64, 2024-06-05:
Tested on BR1 and BR3 here and they both work fine. Perhaps try eraseing before flashing.

pkendall64, 2024-06-06:
More information was added to #2760

closed 2024-06-06 by pkendall64