ExpressLRS/ExpressLRS #2759: Model Configuration file upload fails on DIY 2.4 Generic RX as TX target

Pascal330, 2024-06-04:
## Current Behavior

I flashed a Betafpv ELRS Nano 2.4 Reciever with the latest version of ELRS (3.4.2) and attempted to upload a model configuration file in order to change the default values of packet rate, telemetry ratio and so on without the need of the lua script (soldering it in my Radio as internal module). The WebUi prompted an error message and the settings stayed at default. I even tried uploading a model.json file I had just pulled from the exact same device.

## Steps to Reproduce

1. Let device enter wifi mode (it doesn‘t matter if you connect in AP mode or after connecting the device to your home network)

2. On the main Page of the WebUi klick „upload model configuration file“ and select the file you want to upload

3. In my Case the WebUi prompted me with the error „an error occurred while uploading model configuration file“

[image]

## Possible Solution (Not obligatory)

Maybe the RX as TX target code does not fit the ESP8285 hardware built into ELRS recievers?

## Details

I am trying to use a Reciever as small alternative to a transmitter on a Arduino based Radio Controller. Before using it I would like to change the default values of Packet rate, telemetry ratio, switch mode, model match and power output mode. Opening up my radio and soldering the device to the internal module seems a little extreme for tweaking some settings. Especially since the Configuration upload setting already exists.

## Your Environment

- TX hardware: Betafpv Elrs Nano RX 2.4
- RX hardware: Betafpv Elrs Nano RX 2.4
- Handset model: Frsky QX7S / Arduino Ble nano 33 Rev 2
- EdgeTX version (2.8.2)
- ExpressLRS version (3.4.2)
- MacBook Air 2019 with google Chrome / Iphone X with safari

CapnBry, 2024-06-11:
RX as TX is incredibly RAM limited and craps out. It can't even generate the model config file correctly on current releases. It might "just work" after the changes to the ArduinoJson library/code in #2732 though? Try flashing master

Pascal330, 2024-08-03:
Hey Hey there, I just saw that the issue is closing, probably because there has been a recent update. I tried the same method again to upload the model configuration file via the Web Gui (after updating the Firmware) but sadly there was no change in the behavior. The File only contains one line of parameters. The prompt came immediately after choosing the configuration file on my desktop.

pkendall64, 2024-09-22:
Theres just not enough free RAM to handle this on the ESP8285.

closed 2024-09-22 by pkendall64