ExpressLRS/ExpressLRS #2760: Issues with BR1/BR3 not entering WiFi mode quickly after updating to 3.4.2.

zyren, 2024-06-05:
Issues with BR1/BR3 not entering WiFi mode quickly after updating to 3.4.2.

Since the previous issue report #2757 was closed, I have continuously tried various methods and finally concluded, kindly requesting the development team to review this issue description again and help resolve it, thank you very much.

Method 1:
···
1. Update BR1/BR3 firmware to 3.4.2, selecting Erase before updating.
2. Power on the receiver, the LED blinks twice for binding mode, press the receiver button for 5 seconds, the LED rapidly flashes for WiFi mode, but the SSID cannot be found.
3. Bind, then activate RX WiFi via LUA script, the receiver LED rapidly flashes for WiFi mode, but the SSID cannot be found.
4. Turn off the transmitter, and reboot the receiver, press the receiver button for 5 seconds to enter WiFi mode, the SSID: ExpressLRS RX can be found and used normally.
5. Thereafter, whether through Lua, pressing the button for 5 seconds, or waiting 60 seconds, the SSID can be found and used normally.
···

Method 2:
···
1. Update BR1/BR3 firmware to 3.4.2, selecting Erase before updating.
2. Similar phenomena as in Method 1, whether pressing the button for 5 seconds or enabling WiFi in Lua after binding, the SSID cannot be found.
3. Power on the receiver, the LED blinks twice for binding mode, wait 60 seconds to allow the receiver to automatically enter WiFi mode, the SSID can then be found and used normally.
4. Thereafter, whether through Lua, pressing the button for 5 seconds, or waiting 60 seconds, the SSID can be found and used normally.
···

Conclusions:
···
1. After the first firmware update to 3.4.2 (not yet bound), WiFi mode cannot be activated by pressing the button for 5 seconds, even though the LED rapidly flashes indicating WiFi is activated.
2. After the first firmware update to 3.4.2 (already bound), WiFi mode cannot be activated through Lua, even though the LED rapidly flashes indicating WiFi is activated.
3. After the first firmware update to 3.4.2 (whether bound or not), without turning on the transmitter, just power the receiver and wait 60 seconds to enter WiFi mode, it can be used normally, and from then on, WiFi can be correctly activated and used in any way.

I am not sure if this qualifies as a bug, but I think it has caused some confusion for certain users. After the first update to 3.4.2, they may feel frustrated by not being able to easily activate WiFi and I hope this issue can be resolved.
···

pkendall64, 2024-06-06:
The other ticket was not closed, it was labeled "closing" while we investigate and gather more information. But since you've given more information we will close the other ticket now.

These should not be specific to BR1/3 as these use the same firmware as all the other ESP8285 based receivers.

Method 1, step 2. I can see the wifi appear.
Method 1, step 3. Bind, start wifi on RX from Lua, I can see the wifi appear.

All ways to get into wifi appear to be working for me, on the BR3 that I have and on other receivers too.

knoppixmeister, 2024-06-27:
I have my BR3 v1.0. With firmware to 3.4.3 and 3.4.2 as well 3.4.0 its not possible to get WiFi into work.
But looks it perfectly works with ELRS fw. v3.3.0 and it was installed from factory.

BTW, BR3 works only with FCC 915Mhz region/spec/mode or it also possible to run it in another modes like INT_866?
I tried to run it in IN_866 mode on fw 3.4.3 and it looks just hangs after reboot. Led has solid but not strong (faded) light.

pkendall64, 2024-07-20:
A faded LED is a bad flash. You will need to follow the recovery procedure by flashing with an FTDI dongle.
Once flashed you can enter the web UI and change the regulatory domain.

closed 2024-08-02 by github-actions[bot]

Snippy515, 2024-09-11:
I'm actually experiencing the same issue, except I'm not able to connect to the WIFI on any device (I've tried 2 computers and 2 phones. Both have the error of "can't connect to network"). Most times, the SSID doesn't even come up. 

Initially, when it was in factory v3.0.0, I had no problems connecting and updating (I have done about 4 other ELRS RXs through this same method). But after updating to v3.4.3, I'm having WIFI issues.

The biggest headache was that I plan to use this RX for ELRS airport through ArduPilot. But since I can't get into the WIFI to tick the box, I'm stuck.