ExpressLRS/ExpressLRS #2761: WiFi not working in 3.4.2? (Happy Model ES24TX Pro)

psfales, 2024-06-05:
## Current Behavior

I'm unable to access the WiFi AP after upgrading to 3.4.2.   I  haven't had any trouble in the past, including the recent upgrade from 3.4.0 to 3.4.1.

## Steps to Reproduce

1.  Enable WiFi on the HM ES24TX pro.  (Either using the LUA script, or waiting for the 60 second timeout)  LED color changes from "Fading Blue" (No connection to receiver, color indicates packet rate) to "Green Heartbeat"  (Wifi) 

2.  Connect to the "Expresslrs TX" access point and flash 3.4.2

3.  Attempt to enable WiFi on the HM ES24TX pro.  (Either using the LUA script, or waiting for the 60 second timeout)  LED color changes from "Fading Blue" (No connection to receiver, color indicates packet rate) to "Green Heartbeat"  (Wifi) 

4. No "Expresslrs TX" access point is available.  (I've checked phone, tablet, and laptop).   The TX appears to be working otherwise normally, as the LUA script works, and I'm able to connect to a receiver and operate the aircraft.   It is not possible to connect to the internal web server, and not possible to update firmware over wifi.  

## Possible Solution (Not obligatory)

Allow Wifi use as in earlier releases.

## Details

It's not easy to revert and try again, as it requires disassembling the transmitter module and re-loading via the UART, so I would prefer to just stay on 3.4.2 until forced to upgrade for some other reason.  

## Your Environment

- TX hardware:  Happy Model ES24 TX Pro
- RX hardware: NA
- Handset model: Taranis QX7 with inverter mod
- OpenTX version (including nightly number)  2.3.15
- ExpressLRS version (TX & RX MUST MATCH):  3.4.2
- Packet Rate:  333
- Telemetry Ratio:  1:4
- user_defines:  NA

m0p3d, 2024-06-06:
U forgot to mention its most likely FrSky Ethos Platform related - isnt it ?

psfales, 2024-06-06:
I don't know what that means, but I don't think it's related to the handset.

kvcuijck, 2024-06-08:
Although not very familiar with EdgeTX and TX hardware I was upgrading the ELRS module of my sons Jumper T-Lite V2 from a 2.x.x version to the latest 3.4.2. 
After the upgrade I could still start the webserver on the transmitter and connect to the AP (named "ExpressLRS TX"). My PC receives a valid IP address and I can ping 10.0.0.1 which is the ELRS module. In the browser when visiting 10.0.0.1:80 the connection was refused. An NMAP scan resulted in one open port being port 80. To me it seems the webserver on the ELRS module does not start correctly or there is another reason why connection is refused using Firefox and Chrome.

Just tried Microsoft Edge and that browser does seem to work......

psfales, 2024-06-08:
Thanks for the data point.   I kind of figured it couldn't be impacting everyone because it would have a lot more visibility.   I don't know if it's related to my  specific hardware, or some procedural error on my part, or something else.  I'm hoping that someone with the Happy Model ES24TX Pro  can try it and report success/failure.

psfales, 2024-06-12:
Well, my face is red.   I tried it multiple times with multiple devices (phone, tablet, and laptop) over a period of several days and it never worked.   Today I decided to try it one more time before taking it apart, and everything worked fine.   The phone, tablet, and laptop are all able to see the "ExpressLRS TX" access point and connect to it.   

I'm unable to see anything different between what I did before and what I'm doing now, but it's clearly working now. Closing the issue!

closed 2024-06-12 by psfales

psfales, 2024-06-28:
So, I wanted to upgrade from 3.4.2 to 3.4.3 and ran into the same situation.    I'm not going to re-open the issue (it seems like it must be something on my end, but I wanted to document the problem. )   The TX "thinks" it is going into AP mode:  it says so on the LUA screen, and the LED color changes.  But the AP is not visible to on the network (tried several different devices) so it's not possible to do the update.

Fazented, 2024-08-25:
I have the exact same issue, cant see the AP on any device, even though it is running and the LED on the back indicates it is in wifi mode. This is right after updating to 3.4.3, so I might try downgrade it to see if I can get it working again.

I'm using the same Happymodel ES24 TX Pro on a TX16S MK1

psfales, 2024-09-06:
FWIW, I observed this problem after upgrading from 3.4.1 to 3.4.2 (though it worked a few weeks later) and then after upgrading from 3.4.2 to 3.4.3.  I feel that "something" more is going on than just the software version, but I have no idea what that might be.

psfales, 2024-09-24:
As another data point, I was actually able to upgrade from 3.4.3 to 3.5.0 over Wifi.   I'm not exactly sure how I got it to work.  I tried the different methods for activating wifi (waiting for the timeout, and activating from the lua script).  I probably tried each method at least 5 times, both repeatedly  and alternating between them.  At some point, it "just worked" and I was able to see the AP and upgrade to the new version.