ExpressLRS/ExpressLRS #2766: RX produces sporadic pulses on PWM signal line on startup

neoxic, 2024-06-08:
## Current Behavior

When a Radiomaster ER3C-i receiver (UNIFIED_ESP8285_2400_RX_3.3.0) is powered on, it produces invalid output (sporadic pulses and noise) on the line for about 240ms before valid servo PWM output begins.

[image]

## Steps to Reproduce

1.

2.

3.

4.

## Details

The current behaviour constitutes invalid servo PWM input, hence it may lead to all sorts of troubles with dependent gear.

## Your Environment

- TX hardware: Radiomaster MT12
- RX hardware: Radiomaster ER3C-i
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): 3.3.0
- Packet Rate:
- Telemetry Ratio:
- user_defines:

pkendall64, 2024-06-08:
This is a known issue with ESP8285 based receivers. The pins that are used also double as Serial IO pins i.e. the pins use to update the receiver. The problem is that the ESP8285 always outputs serial information at 74880 baud on the TX pin and some other pins have PWM signals during boot.
There is nothing the dev team can do about this as it done by the ESP8285 ROM.
A better receiver for PWM is the ESP32 based receivers as these generally have more pins available and most of the pins are doing crazy boot-time things. The exception is still the TX pin, but that is usually either not used as a PWM pin or is much further along the PWM list.

neoxic, 2024-06-08:
Awww, that's unfortunate. I will need to figure out a logical workaround on my side. Just curious... Would that have been possible to circumvent by using a different pinout? The ESP32 has a GPIO multiplexor, so a timer can be mapped to a different pin. Does the same hold true for the ESP8285?

pkendall64, 2024-06-08:
No the 8285 is pretty limited in what can be used on what pins. So with the SPI taken by the semtech chip and needing 1 and 3 for the serial for updating and 0 for the boot theres very little left. So we had to use those for PWM just to be able to get 5 PWMs available.
With the new C3 support you will get better PWM receivers coming soon I would imagine. These are ESP32 and have a lot of the hardware modules on-board including hardware PWM generation which we use, whereas the ESP8285 is timer based and has a bit of jitter.

neoxic, 2024-06-08:
I see. Well, I only wish that PWM pin on the ESP8285 would have been chosen to be the **last** channel, not the **first** one. At least, that wouldn't be sticking out so bad.😂 But... yeah, we have what we have now. Thanks for the info!

closed 2024-06-08 by neoxic