ExpressLRS/ExpressLRS #2769: ESP32 C3 RX 5PWM

norulers, 2024-06-09:
## Current Behavior

Setting 16CH/2 using LUA script will result in an error

## Your Environment

- TX hardware:Jumper T15
- RX hardware:DIY ESP32-C3 RX
- Handset model:
- OpenTX version (including nightly number)EDGETX 2.11.0
- ExpressLRS version (TX & RX MUST MATCH):TX:3.4.2     RX:main
- Packet Rate:333Hz
- Telemetry Ratio:1:2
- user_defines:
- 
[screen.zip](https://github.com/user-attachments/files/15750800/screen.zip)

pkendall64, 2024-06-09:
That is normal behaviour. It's telling you that you cannot change the switch mode while connected to the RX. Do it before powering up the RX.

norulers, 2024-06-11:
> 这是正常现象。它告诉您在连接到 RX 时无法更改开关模式。请在启动 RX 之前执行此操作。

> That is normal behaviour. It's telling you that you cannot change the switch mode while connected to the RX. Do it before powering up the RX.

Thank you for your reply. The settings are valid after disconnecting the RX connection

norulers, 2024-06-11:
> That is normal behaviour. It's telling you that you cannot change the switch mode while connected to the RX. Do it before powering up the RX.

Thank you for your reply. The settings are valid after disconnecting the RX connection