ExpressLRS/ExpressLRS #2778: Flysky TX Modules Support

richardclli, 2024-06-13:
The following Flysky TX modules will be supported:

1. IRM301 The internal module of Flysky EL18 radio
2. FRM303 The external module by Flysky

Will work with Flysky to develop a port of ELRS firmware which is compatible with their stock TX bootloader, so that they can provide a tool to flash their TX modules to ELRS firmware, or flash back to their stock AFHDS3 firmware.

JyeSmith, 2024-06-13:
@richardclli please use our current discord group chat to discuss.  As you are aware ELRS is not looking to add or support new STM hardware.  STM support will be removed in the future.

closed 2024-06-13 by JyeSmith

pfeerick, 2024-06-13:
Thanks for putting this up Richard. It will be good to see the internal RF on the EL18 having an ELRS option, as well being able to use the higher power FRM303 module for that also.

richardclli, 2024-06-14:
> @richardclli please use our current discord group chat to discuss. As you are aware ELRS is not looking to add or support new STM hardware. STM support will be removed in the future.

Yes, understood, however we can still have some fun in the meantime ^.^

richardclli, 2024-09-27:
It is done, check it out int he following link:

https://github.com/richardclli/Flysky-ELRS