ExpressLRS/ExpressLRS #2783: no such file or directory

Rothammel, 2024-06-15:
## Expected Behavior

flash bootloader

## Current Behavior

FileNotFoundError: [Errno 2] No such file or directory: 'bootloader/r9m_lite_pro_bootloader.bin'

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Context

first try

## Your Environment

Windows 11

```
 * ST-Link/V2 V2J31S7 is not recent firmware, please upgrade first - functionality is not guaranteed.

DEVICE: ST-Link/V2 V2J31S7
SUPPLY: 3.15V

CORE:   CortexM4

MCU:    STM32F302xC/STM32F303xC/STM32F358xC
FLASH:  256KB

 * ST-Link/V2 V2J31S7 is not recent firmware, please upgrade first - functionality is not guaranteed.

DEVICE: ST-Link/V2 V2J31S7
SUPPLY: 3.16V

CORE:   CortexM4

MCU:    STM32F302xC/STM32F303xC/STM32F358xC
FLASH:  256KB
Traceback (most recent call last):
  File "runpy.py", line 196, in _run_module_as_main

  File "runpy.py", line 86, in _run_code
  File "C:\Users\stan\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\2d98919b5a4c780d3d1c8400dd478ef99a4b3025\firmware\flasher.pyz\__main__.py", line 3, in <module>

    if __name__ == '__main__':
  File "C:\Users\stan\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\2d98919b5a4c780d3d1c8400dd478ef99a4b3025\firmware\flasher.pyz\_bootstrap\__init__.py", line 253, in bootstrap
  File "C:\Users\stan\AppData\Roaming\ExpressLRS Configurator\firmwares\cloud\ExpressLRS\2d98919b5a4c780d3d1c8400dd478ef99a4b3025\firmware\flasher.pyz\_bootstrap\__init__.py", line 38, in run
  File "C:\Users\stan\.shiv\flasher.pyz_b72927cdf2af60919aae2889f98a80edfcb28d05c345ca5f6c907f4d429f2984\site-packages\binary_configurator.py", line 413, in main

    return binary_flash.upload(options, args)
  File "C:\Users\stan\.shiv\flasher.pyz_b72927cdf2af60919aae2889f98a80edfcb28d05c345ca5f6c907f4d429f2984\site-packages\binary_flash.py", line 197, in upload
    return upload_stm32_stlink(args, options)
  File "C:\Users\stan\.shiv\flasher.pyz_b72927cdf2af60919aae2889f98a80edfcb28d05c345ca5f6c907f4d429f2984\site-packages\binary_flash.py", line 65, in upload_stm32_stlink
    stlink.program_flash('bootloader/' + options.bootloader, flash_start, erase=True, verify=True, initialize_comms=True)
  File "C:\Users\stan\.shiv\flasher.pyz_b72927cdf2af60919aae2889f98a80edfcb28d05c345ca5f6c907f4d429f2984\site-packages/external\pystlink\pystlink_api.py", line 257, in program_flash
    mem = self._read_file(str(firmware))
  File "C:\Users\stan\.shiv\flasher.pyz_b72927cdf2af60919aae2889f98a80edfcb28d05c345ca5f6c907f4d429f2984\site-packages/external\pystlink\pystlink_api.py", line 283, in _read_file
    with open(filename, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'bootloader/r9m_lite_pro_bootloader.bin'

```

jurgelenas, 2024-06-15:
Hey @Rothammel 

Appologies for missing your issue. I have transferred it to the main repo. It looks like a firmware issue, not the configurator.

pkendall64, 2024-06-16:
@Rothammel can we have more information please, like what version of firmware were you flashing?

pkendall64, 2024-06-16:
Never mind. I could see from the hash that was version 3.3.2. I have updated the artefact in our artefact repository so this is now fixed.

closed 2024-06-16 by pkendall64