ExpressLRS/ExpressLRS #2784: LUA Set Failsafe function doesn't respect Rx input/output channel mapping on pwm receivers

AlfieLockrey, 2024-06-15:
## Current Behavior

Using a Matek 6 channel PWM receiver where the outputs 1-6 have been mapped to inputs 6-1 in order to reverse the physical ordering of the servo pins. 

Setting failsafe to current position to obtain trimmed flight in the evening of failsafe using the Lua script on a boxer running latest 2.10 edge tx and latest Lua.

## Steps to Reproduce

Manually setting the failsafe in the web UI works (set output 6 (ch3) to 988) and throttle drops on motor when Tx turned off.

However when setting failsafe in Lua on Tx (in order to failsafe to a trimmed servo position and 0 throttle), output failsafe values are mapped to the same number channel input (output 1 FS value = ch1 current value).

This is presumably not expected behaviour and would result in a flyaway for this edge case of re-ordering channel I/O on the receiver side.

## Possible Solution (Not obligatory)

From what I can understand this issue arises from the rxDev_lua.cpp file with the luaparamSetFalisafe function (not my typo) where rx outputs n to m are mapped from the channel inputs n to m, disregarding any I/O mapping.

I was hoping I could report the issue and offer a code change at the same time but wanted to make the issue before I forgot as I am away from my pc for a while.

I have been unable to test with other hardware but will re-verify when able to with what I have available.

pkendall64, 2024-06-17:
I have just tested this on a BETAFPV SuperP 14 RX. I switched the order of pins 1-5 so pin 1 is channel 5 , 2 is 4 etc,.
I then entered the Lua script and held pitch (channel 2) high and arm was low (channel 5) and set failsafe.
I then checked in the webUI that indeed it had respected the mapping.
Then went on to test the failsafe. I set arm HIGH, and left pitch in the middle then powered off the radio. The channel 2 (pin 4) moved to the 2000us (high position) and the channel 5 (pin 1) moved to the 0us (LOW) position.

This is working as expected.

AlfieLockrey, 2024-06-17:
Interesting, I unfortunately left the Rx with a friend by accident but think I have another, I will test on their current firmware again and update and test again.

Thanks for testing,
Alfie

AlfieLockrey, 2024-06-23:
Got the reciever back in hand and tested, it wasn't a matek but a happymodel EPW6. The issue was present in the older firmware but since updating to 3.4.2 the issue appears to be fixed and I hadn't seen the PR fixing this.

closed 2024-06-23 by AlfieLockrey

pkendall64, 2024-06-23:
> Got the reciever back in hand and tested, it wasn't a matek but a happymodel EPW6. The issue was present in the older firmware but since updating to 3.4.2 the issue appears to be fixed and I hadn't seen the PR fixing this.

It may have been fixed as a side-effect of other work. Thanks for verifying.