ExpressLRS/ExpressLRS #2790: FC to Tx telemetry stops after a while

rotorflight, 2024-06-17:
## Current Behavior

The custom telemetry frames send by Rotorflight are not send to the Tx after a while.
Sometimes it's just a few minutes, sometimes a few hours.

Power cycling the Rx fixes the issue. Power cycling Tx or FC does not help.

## Steps to Reproduce

Send custom CRSF frames from FC to Tx (0x88 in Rotorflight, with varying size).

Faster frame rate makes it happen sooner.
Generally it does not matter if the frames are send faster or slower than the Rx can transmit them.

## Details

Unfortunately there are no errors shown. More details will be added later, if anything comes up.

## Your Environment

- TX hardware: RM Ranger Micro
- RX hardware: RM RP-3H
- Handset model: RM TX16S
- OpenTX version: EdgeTx 2.9.4
- ExpressLRS version: TX 3.3.1 + RX tested 3.3.1, 3.3.2, 3.4.1, 3.4.2
- Packet Rate: 500
- Telemetry Ratio: 1:2 ... 1:16
- user_defines: Nil

mha1, 2024-07-15:
can you please test https://github.com/ExpressLRS/ExpressLRS/pull/2834

rotorflight, 2024-07-16:
I managed to narrow down the problem.

We had a bug in our code, which caused the CRSF frame format to be incorrect.
Surprisingly, most frames still got through to the Tx, so it took a while to understand what was happening.
Essentially, we did not have the source and destination addresses in the frame:
```
0xC8   SYNC
0xXX   Frame length
0x88   Frame type = RF custom telemetry
....   Data
```

Once we fixed it to the correct format, the lockups stopped:
```
0xC8   SYNC
0xXX   Frame length
0xEA   Destination = Tx
0xC8   Source = FC
0x88   Frame type = RF custom telemetry
....   Data
```

As the src and dst fields had "random" data in them, it's ok that things were not 100% working.
But I suspect the lockup is still a bug in the Rx, that just got triggered by the "unusual" testing.

The lockup has not happened since the frame format was fixed.

rotorflight, 2024-07-16:
The lockup is not related to #2834.

mha1, 2024-07-16:
are we good to close this issue?

rotorflight, 2024-07-16:
Is it ok for the Rx to lock up if it receives malformed frames. 
Or is it a bug if it locks up in that situation.
That is the question.

JyeSmith, 2024-07-16:
> Is it ok for the Rx to lock up if it receives malformed frames. Or is it a bug if it locks up in that situation. That is the question.

Malformed packet should be caught here.

https://github.com/ExpressLRS/ExpressLRS/blob/fa89fdd5ce701c8ef4de3212a04c14d8e70ea1e2/src/lib/Telemetry/telemetry.cpp#L166

Are you able to do another log with the malformed packet and capture the TLM packet sent to the Rx before the lockup?  It might appear to be a CMD to restart in bootloader mode.

https://github.com/ExpressLRS/ExpressLRS/blob/fa89fdd5ce701c8ef4de3212a04c14d8e70ea1e2/src/lib/Telemetry/telemetry.cpp#L244

rotorflight, 2024-07-17:
I will write some test code to capture the packet that is causing it. Assuming I can get it to lock up again.

pkendall64, 2024-08-02:
@rotorflight did you manage to reproduce this again? Or are you happy that this can be closed?

closed 2024-08-31 by github-actions[bot]