ExpressLRS/ExpressLRS #2796: Reminder to add BLE Joystick shortcut on the customizable buttons on the Ranger, Bandit etc.

NorgeSquad, 2024-06-19:
^

JyeSmith, 2024-07-14:
@NorgeSquad can you please test this PR https://github.com/ExpressLRS/ExpressLRS/pull/2842

NorgeSquad, 2024-07-15:
Hi! Yes sure! Im not that good technical. so where can I find the 
download? :)

mha1, 2024-07-15:
It got merged into master. Just use the Configurator with tab GIT BRANCH selected to build branch master. If you don't see the tab GIT BRANCH go to Settings and check Expert Mode.

NorgeSquad, 2024-07-15:
Allrighty. Got it! But whats the branch called in the Git branch?

mha1, 2024-07-15:
master

NorgeSquad, 2024-07-19:
Okay nice! My bagagde got lost at the airport with my Radio and module. 
But when I get it I will let you know. But Seems like it worked great!

NorgeSquad, 2024-07-23:
Tested it now! Worked great! Would you guys recommend me to stay in the 
Master Git branch. Or should I flash back to Newest official Firmware?

mha1, 2024-07-23:
I'd stay on the latest official firmware and wait fir 3.5 to be released

NorgeSquad, 2024-07-24:
Oh Alright! I assume 3.5 is Around the corner

NorgeSquad, 2024-07-24:
I found out that the BLE joystick Timeout every 20th minute. is there 
any way around this?

JyeSmith, 2024-09-22:
Fix here https://github.com/ExpressLRS/ExpressLRS/pull/2944

closed 2024-09-22 by JyeSmith