ExpressLRS/ExpressLRS #2805: DIY Mini ESP32 SX1280 2.4GHz TX hangs on loading LUA script in versions 3.4.0, 3.4.1, 3.4.2

wchnflr, 2024-06-21:
## Current Behavior

I flashed a DIY Mini ESP32 SX1280 2.4GHz TX to 3.4.0, 3.4.1, and 3.4.2, but the LUA script from the ELRS Configurator doesn't load in any 3.4.x firmware. The LUA script from 3.3.2 also doesn't load.
I can downgrade to 3.3.2 and everything works there.

## Steps to Reproduce

[image]

I used the same settings that worked from 3.3.2 for 3.4.0, 3.4.1, and 3.4.2.

## Details

I was attempting to upgrade from 3.3.2 to 3.4.x since the release notes recommend that everyone should upgrade with no downsides.

Is there a way I can debug this myself? Where in the source code should I start?

## Your Environment

- TX hardware: DIY Mini ESP32 SX1280 2.4GHz TX from https://www.expresslrs.org/hardware/special-targets/nuclear-hardware/
- RX hardware: N/A
- Handset model: TBS Tango II PRO
- OpenTX version (including nightly number): fdtx-tango TBS-1.3.5 2022-05-10 EEPR: 219
- ExpressLRS version (TX & RX MUST MATCH): 3.4.0, 3.4.1, 3.4.2; RX version N/A
- Packet Rate: N/A
- Telemetry Ratio: N/A
- user_defines:

pkendall64, 2024-06-21:
What baud rate do you have set to connect to the module? It should really be set to something like 921k or higher if possible.
There are some changes coming with some optimisations in 3.5. If you would like to test them they are in PR #2772

wchnflr, 2024-06-21:
> What baud rate do you have set to connect to the module? It should really be set to something like 921k or higher if possible. 
I don't know at the moment since the handset doesn't provide this info. The only thing I found was this discussion in the freedomtx repository saying it's 400Kbaud, so this what I have to go with:
[https://github.com/tbs-fpv/freedomtx/issues/12#issuecomment-741486706](url)

> There are some changes coming with some optimisations in 3.5. If you would like to test them they are in PR #2772
I'm trying to set up my toolchain to test. Vscode/platformio are both new to me, and it's not going as smoothly as I had hoped.

wchnflr, 2024-06-22:
> > There are some changes coming with some optimisations in 3.5. If you would like to test them they are in PR #2772
> 
> I'm trying to set up my toolchain to test. Vscode/platformio are both new to me, and it's not going as smoothly as I had hoped.

After fumbling through the dark, I finally managed to upload "Unified_ESP32_2400_TX_via_WIFI" (there was no project task available for any "DIY_2400_TX_*") using the changed files in PR #2772. It fixed my LUA loading problem, thank you!

As a bonus, my tx power is no longer capped at 25mW, although that's not always a safe feature since my internal SX1280 module has no heatsink nor cooling fan.

closed 2024-06-23 by JyeSmith