ExpressLRS/ExpressLRS #2815: ELRS 3.4.x not switching diversity antenna correctly...?

AxelRed65, 2024-06-28:
## Current Behavior

I have a BetaFPV Super D receiver 2.4 GHz, using it in Diversity mode. It appears the receiver always locks on Radio/Antenna 2 and never switches to the other antenna, whatever the level is ( the indication shows even full red -120 on the display )... If I revert back to ELRS version 3.3.2 the Antenna/Radio switching performs flawless

## Steps to Reproduce
I have a drone with 2 horizontal dipoles oriented each one perpendicular to the other...
On the radio I get back the RX received RSSI's, RSNR, ANT telemetries shown on a LUA page, also on Googles OSD I get the used radio and its RSSI value...

How to reproduce :

1) On the field, I fly the drone about 500-1000 meters from the radio, hover there and slowly yaw the drone 360° in place, so that both antennas get cross oriented with the radio in turn. On radio and OSD, the antenna indications remain always stuck on Receiver 2 ( sometime it flashes to receiver 1 for few tenths of a second ). Even if stopping the rotation to have antenna 2 the worst position in respect to the radio, nothing changes, it never switches to antenna 1. Reverting to ELRS 3.3.2, same place, same distance, during the rotation I can clearly see the indications of the receiver switching to the better oriented antenna, and the antenna 1 level is well above -60/50 dBm, so, no way that with 3.4.x it should have not switched antenna.

2) In the Lab, I am able to place the Radio ( Jumper T20S ) near under the drone in a way that the drone antenna aligned with the radio saturates, giving out a value around -120dBm, while the other one gets a good level ( about -10/-20 )... 
Orienting the radio correctly I can see the saturation of a single antenna while the other has a good signal, but the reported 'ANT' telemetry parameter is stuck to '1' ( Receiver 2 ), whichever antenna is receiving a good signal. This happens also on the Reported values on the OSD, with a '-120:2' red blinking RSSI value.
If I revert to ELRS 3.3.2 I can clearly see the receiver switching to the unsaturated antenna, both on the Radio and on the OSD

## Possible Solution (Not obligatory)

Check if this is really a bug, or some new feature I found nothing about... Could also be a wrong reported telemetry indication?

## Details

The ELRS FW is always uploaded by ExpressLRS configurator via Passthrough or WiFi using 3.4.0, 3.4.2, 3.4.3 and 3.3.2 release versions.
All versions >= 3.4.0  have the reported issue.

## Your Environment

- TX hardware: Jumper T20S ELRS version 3.4.2
- RX hardware: BetaFPV Super D RX
- EdgeTX 2.10.1 Lua Scripts ELRS 3.0
- ExpressLRS version (TX & RX MUST MATCH): Read above
- Packet Rate : 100Hz  Full
- Telemetry Ratio: 1:32

mha1, 2024-06-28:
This is not a bug, it's a feature.

3.4 changed the the meaning of ANT for true diversity (two independent receiver paths - radios). If a radio was successful in receiving and decoding an incoming OTA packet it pulls its dedicated interrupt line to tell the processor a packet is ready for further processing. If both radios were able to receive and decode the packet both will pull their interrupt lines. The higher priority radio will be used for further processing, the lower priority radio will be discarded. On the SuperP and SuperD radio2 (ANT1) is the higher priority radio.

With having the priorities in mind there are 4 cases:
Both Radio1 and Radio2 were successful -> the higher priority radio's packet (Radio2) will be used -> ANT1 will be reported
Only Radio1 was successful -> Radio1's packet will be used -> ANT0 will be reported
Only Radio2 was successful -> Radio2's packet will be used -> ANT1 will be reported
no radio was successful -> no packet is available -> ANT1 will be reported

So you'll see ANT1 most of the times, ANT0 only if Radio1 was the only successful radio

Takeaway:
- ANT doesn't mean Antenna in case of true diversity receivers
- ANT should have 4 states to reflect what's going on but this would confuse even more users (what? I don't see 4 antennas)

AxelRed65, 2024-06-28:
Thanks for the explanation, but, IMHO there may be a problem, at least with BF 4.5, flying LR FPV...
What I get on the OSD ( Walksnail HD ) is the reported RSSI of Antenna 2 till it gets at least one valid packet, even if its RSSI level is very low, may be even red flashing...
If I need to judge the level of RF reception reported by the drone via OSD, to check if I am reaching a too far low level, how could I...?
In my Lab test ( and on the field ), it reports me always the RSSI Level of radio 2, and if I have only that value, how could I judge if I am reaching my real RF limit ( in diversity mode I mean )...? If I accept a so low RSSI from radio 2 and fly away waiting for Radio 1 and its RSSI to show up, I can not be sure if also Radio 1 is in the same situation as Radio 2, or it may be receiving a good signal...
I think the shown RSSI at least on the OSD should be the better one, so that someone can judge the signal margin it still has, whichever radio is receiving the packets, especially in LR, where you can reach very low RSSI Levels and need to take care of them...
Not sure however if this may depend on ELRS or BF...
Sorry to bother you with my doubts...

pkendall64, 2024-06-28:
You say it should be the showing the better RSSI, other pilots say it should be the worse RSSI because they don't want to lose their craft. They would rather be safe than sorry.
Ideally BF would allow the display of both RSSI values, then the user can make a fully informed decision.

AxelRed65, 2024-06-29:
Don't keep me wrong, I see the point in seeing the worst RSSI, and I would not object to it, even worst RSSI would be fine... What I have doubts about is that you are not looking at best or worst RSSI here, you are looking at the RSSI of Radio 2 ( in this case ) if both radios receive a packet, whatever its level is, worst or best...
Just to be clear, I am no one here, and I do not want to discuss about this feature, if this choice has been done I think it comes from already done discussions and evaluations in the ELRS group...
I was looking into why the radio switching indication was behaving that way, as coming from ELRS release 3.3.x I started thinking if one of the radios of the Super D was having reception problems...and in any case, being ELRS an open project, I can eventually change the code the way I prefer without bothering the developers...

AxelRed65, 2024-06-29:
@mha1 

About this

> * ANT should have 4 states to reflect what's going on but this would confuse even more users (what? I don't see 4 antennas)

Well, why not make the antenna indicator kinda like a 'ranking indicator'...?

'2' means full mode, both radios linked, and then show worst RSSI on OSD, so that one can then judge the signal distance to degraded mode...

'1' Degraded mode, only one radio is receiving packets, show the receiving radio RSSI on OSD, and you are warned that you are already in a non safe state, and you have numbers to decide what to do...

'0' ...FUBAR...

Just an idea...

mha1, 2024-06-29:
Regardless of content one more precious Link Stats Bit will be required. I don't know off hand if that'd lead to spending another byte. My request was frowned upon some time ago.

silicongarage, 2024-07-16:
Not switching to the receiver with better SNR/RQly brings the downside of Dynamic Power switching  keeping both transmitter and receiver telemetry power elevated (potentially much) more than needed.  Not to mention telemetry warnings constantly calling "Wolf" when it's just a positional alignment issue.   If I had a vote, I would favor always switching to the best receiver path (the state of Least Surprises)

kvalexey, 2024-07-21:
I confirm that changing the ANT attribute algorithm is a bad decision for LongRange BF pilots. 
BF4.5 shows in the OSD the RSSI_DBM for the antenna from the ANT attribute, and the dynamic power uses this value. 
For the ELRS 3.3 pilot in the OSD see the best RSSI_DBM and the best antenna number, the dynamic power corresponds to the best DBM_RSSI. 
For ELRS 3.4 pilot see in the OSD RSSI_DBM for the 1st antenna until the 1st antenna can receive the signal, the dynamic power corresponds to the 1st DBM_RSSI. If the 1st antenna is located along the course, the pilot has a worst DBM_RSSI and the dynamic power increases to a bigest value.
After upgrading to ELRS 3.4 , I has 1 W of dynamic power at the same distance, where 50 mW is enough for ELRS 3.3.

mha1, 2024-07-21:
@CapnBry  - Bat-Signal

CapnBry, 2024-07-21:
> Regardless of content one more precious Link Stats Bit will be required. I don't know off hand if that'd lead to spending another byte. My request was frowned upon some time ago.

You can't "add another byte" to OTA, all the bytes are in use and there's no more space available. 

I was unaware of the changes made to how the antenna is reported in 3.4, but if it isn't reporting to the dynamic power algorithm which value to be used for FLRC, the obviously it isn't going to work properly. Also all scripts that rely on it and OSDs. 

I'm not around today but I'm also not sure how this can be addressed if the one bit value isn't representing which value is useful? I should note that as far as "antenna switching", true diversity doesn't switch apart from which antenna is used for telemetry transmission so at least the receiver side is ok.

AxelRed65, 2024-07-24:
Well, I have managed to restore old behavior where the telemetry is reporting the best signal/antenna in the elrs 3.4, easy change in the code, and it works... It's a month I am using it, and no more hot TXer running @ 1W continuously when the drone is about few hundred meters from me just because an antenna is wrongly oriented...
I have not so much time now, but I think I will also find a way to apply my previous idea where I use indication 2 as good on both Antennas, 1 Degraded mode indication... No indication of lost signal, but, not even now it is present, so, nothing would change... but I need to have some free time...

silicongarage, 2024-07-24:
Is this the only change you had to revert? :
[image]

AxelRed65, 2024-07-25:
No, that change is for the receiver... You need to change the processing for the Txer, so that also the OSD appears correct... so, in ExpressLRS\src\lib\SX12xxDriverCommon\SX12xxDriverCommon.h I removed the Fuzzy logic function this way:

>
> 
> int8_t fuzzy_snr(int16_t snr1, int16_t snr2, int16_t threshold)
>     {
> 
>     #if 0
> 
>         if (threshold == 0)
>         {
>             return (int8_t)(snr1 < snr2 ? snr1 : snr2);
>         }
> 
>         int16_t diff = (int16_t)abs(snr1 - snr2) << 4;
>         int16_t lower_value = (int16_t)(snr1 < snr2 ? snr1 : snr2) << 4;
>         int16_t average_value = ((int16_t)snr1 + snr2) << 3;
> 
>         int16_t threshold_scaled = (int16_t)threshold << 4;
> 
>         if (diff < threshold_scaled)
>         {
>             return (int8_t)(lower_value >> 4);
>         }
>         else if (diff > threshold_scaled * 2)
>         {
>             return (int8_t)(average_value >> 4);
>         }
>         else
>         {
>             int16_t transition_value = (diff - threshold_scaled) / threshold_scaled;
>             return (int8_t)((lower_value * (16 - transition_value) + average_value * transition_value) >> 8);
>         }
> 
>     #else
>         return (int8_t)(snr1 >= snr2 ? snr1 : snr2);
>     #endif    
>     }

silicongarage, 2024-07-29:
update:  I patched V3.4.3 with the above revert and have now flown over 1.5 hours with no problems.  TX has returned to the expected low power behavior.  Comparing a before/after flight,  TX MAX power states have dropped from 35% to 1%.  Also significant improvements in Avg RSS, RSNR, TRSS, TSNR

anti-vaxxer, 2024-09-20:
I experienced this problem too after updating to 3.4, using 2.4G EP Dual diversity RX, dynamic power was increasing to max several times. I wasn't even far away, it doesn't make sense, this didn't happen before updating.

silicongarage, 2024-12-06:
Fully agree that the antenna signal path selection algorithm  (`Stuck antenna mode` or `Best RSSI mode`) should be an option either in the configurator.app at build time or in the tx lua script.

silicongarage, 2024-12-16:
The misplaced belief of _extra margin_ from `stuck_mode` only works half the time.  

If ANT1 is mis-oriented to start with, then yes there will be margin to fall back on when it finally _unsticks_ and falls back to ANT0.   However if ANT1  starts out in optimal alignment , then when it finally drops out there is nothing to fall back on.  Blindly depending on `stuck_mode` to provide safety margin will lead to a rude surprise.  

Unfortunately, there is no way to tell which state of alignment ANT1 maybe in.   Are you going to be in for a surprise when ANT1 drops out?     **¯\\\_(ツ)\_/¯** 

This is the one of the reasons (dynamic power being another) that the receiver should always choose the `best RSSI` antenna.  Telemetry will  be reporting the actual state of signal strength and informed decisions can then be made without surprises.

GiveDaData, 2025-09-17:
for fixing the dynamic power, instead of reverting the antenna code in rx_main.cpp
it might be better to just change the rssi values used by DynamicPower_Update in dynpower.cpp

currently, it uses the active antenna rssi value. 

int8_t rssi = (CRSF::LinkStatistics.active_antenna == 0) ? CRSF::LinkStatistics.uplink_RSSI_1 : CRSF::LinkStatistics.uplink_RSSI_2;

If anyone wants the dynamic power to act according to the best antenna rssi value, then change it to

int8_t rssi = (CRSF::LinkStatistics.uplink_RSSI_1 >= CRSF::LinkStatistics.uplink_RSSI_2) ? CRSF::LinkStatistics.uplink_RSSI_1 : CRSF::LinkStatistics.uplink_RSSI_2;

CRSF::LinkStatistics.uplink_RSSI_1 and CRSF::LinkStatistics.uplink_RSSI_2 are negative numbers