ExpressLRS/ExpressLRS #2838: Heatsink gets hot while just using BLE joypad on T-Pro V2 900mhz

thelabexpedition67, 2024-07-13:
## Current Behavior

I noticed that when I'm using the BLE joypad on the Jumper T-Pro V2 (900mhz version) the heatsink on the back gets hot.

## Steps to Reproduce

1. Power on the radio

2. Select the BLE joypad from the lua tool and connect to pc

3. Using it for some time

4. The heatsink gets hot

## Details

I described the issue on the discord channel and a theory is that the internal AMP could stay on when the BLE joypad is used, this will drain the battery faster and I noticed also that after some time I get some signs of unresponsiveness as well. 

I measured the external temperature of the heatsink and it is reaching about 55°C (Ambient at 25°C).

I really do not know if it is something related to the firmware, I hope so.

## Your Environment

- TX hardware:  Internal ELRS 900Mhz 
- Handset model: Jumper T-Pro V2
- EdgeTX version:  2.10.1 (Centurion)
- ExpressLRS version: 3.4.2 EU868

pkendall64, 2024-07-13:
Going through the code there are a few things done when entering BLE mode.
1. The output power is set to minimum power level
2. The Semtech chip is set to standby mode
3. The power amplifier is set to disabled/standby mode

So, the only conclusion I can come to is that there is something else going on.

JyeSmith, 2024-07-14:
it the RF amp that Jumper is using.  The quiescent current at 5V is 260mA, over 1W while idle.

closed 2024-07-14 by pkendall64