ExpressLRS/ExpressLRS #2849: How to bind TX and RX without  remote?

petrenkodesign, 2024-07-17:
### Discussed in https://github.com/ExpressLRS/ExpressLRS/discussions/2844

<div type='discussions-op-text'>

<sup>Originally posted by **petrenkodesign** July 14, 2024</sup>
I use the TX Happy Model ES900 Max (I suspect that this is some kind of ES900 clone, but very high-quality), drone has an ES900RX. Flashed both modules with the latest release 3.4.3, and added a passphrase. When the TX is in the remote control, modules are connected without problems. As soon as I try to connect the modules, when the TX is separate from the remote control, nothing happens, the modules are in the search and after 60 seconds they switch to wifi mode. I sent a binding command via UART (0xEE, 0xEA, 0xFF, 0x00), nothing changes. What am I doing wrong?</div>

JyeSmith, 2024-07-17:
ELRS does not connect OTA without a crsf signal e.g. with no handset connected.  This is because there is no RC data to send.

The HM ES **Max** is also unsupported by ELRS.

petrenkodesign, 2024-07-18:
> ELRS does not connect OTA without a crsf signal e.g. with no handset connected. This is because there is no RC data to send.
> 
> The HM ES **Max** is also unsupported by ELRS.
Already supported :)
https://github.com/ExpressLRS/targets/compare/master...petrenkodesign:elrs_targets:HM_ES900_TX_MAX
As I wrote earlier, when the module is in the remote control, the connection is made, and the drone is controlled normally.

I transmitted signal read from RX CRSF via UART to TX, also sent a request to bind (0xEE, 0xEA, 0xFF, 0x00) it did not work.
Maybe there is some sequence I'm missing?

pkendall64, 2024-07-20:
You do not need to send a binding command. You need to continuously send RC commands to the TX module.
See https://github.com/crsf-wg/crsf/wiki

Also converting this to a discussion as this is not a bug.