ExpressLRS/ExpressLRS #2855: Radiomaster ER5Av2 no Reaction to Team Race Settings

r41065, 2024-07-22:
Radiomaster ER5Av2 shows no Reaction to Team Race Settings, getting Telemetrie and Servo Movements at all possible selected settings. ( no Failsafe Reaction at not selected positions )
Same Settings on Radiomaster ER6 work fine.

## Steps to Reproduce
Select Team Race , select Aux Channel (tested 7..12), set Position to one of the 6 possibe.

- TX hardware: DIY 2400E28
- RX hardware: RadioMaster ER5Av2
- Handset model: FrSky X9Lite / RadioMaster Zorro
- OpenTX version (including nightly number) EdgeTX2.10
- ExpressLRS version (TX & RX MUST MATCH): 3.4.0RC2 ... 3.4.3
- Packet Rate: 100Hz Full
- Telemetry Ratio: 1/64
- user_defines: elrsV3.Lua

pkendall64, 2024-12-30:
I thought I'd do some investigation into this issue, and it appears that TeamRace is really only supported on serial connections for flight controlled devices. It just happens to do some of the work related to TeamRace on the the ER6 because it has a serial port defined by default, whereas the ER5 does not. Hence why the servos don't actually go into failsafe position when using the ER6.
If you can give us a use case as to what you are using the team race feature for with PWM controlled models, then perhaps we can create a feature request issue for it.

closed 2024-12-30 by pkendall64

r41065, 2024-12-30:
Im building Ship Models with many Functions like SAR.
For example Here you can use it to drive the dinghy seperaty to the main Model. In an other case the Crane Function is seperate ( with no physical Connection to the Main Model Ship ,)
The Interesting Part would be that you then can use Telemetry from both just by switching this Function.

pkendall64, 2024-12-30:
I see, that's quite an interesting use case. I'll see what I can to support this on PWM receivers. It's will probably be for the 4.0 release though, so that may be some time away.