ExpressLRS/ExpressLRS #2859: elrsv3.lua: support string parameter type (0x0a)

wimalopaan, 2024-07-25:
Using the string parameter type gives an error of `elrsv3.lua`.

This type of parameter is documented in https://github.com/crsf-wg/crsf/wiki/CRSF_FRAMETYPE_PARAMETER_SETTINGS_ENTRY
but it looks that it is not supported in `elrxv3.lua`.

pkendall64, 2024-07-25:
Can you explain why the ELRS lua needs to implement the String parameter type? ExpressLRS itself does not have any string parameters.

wimalopaan, 2024-07-25:
I need it for my own kind of flight controler.
I came across this documented parameter type and I would be glad using it. But I also see the difficulty to implement an edit-box on the script side.

pkendall64, 2024-07-25:
The problem we have is that the current Lua is at the edge of how much free memory is left on EdgeTX.
The only way to solve this is to break the Lua up into chunks for which means it's a multi-file upload to the SD card and users have a hard enough time uploading this one file 😓

wimalopaan, 2024-07-26:
Well, maybe it could be integrated into EdgeTx as C++, e.g. like Openi6x this has done?

wimalopaan, 2024-07-26:
Just tested it with Openi6x, which simply displays a string (0x0a) parameter, but has no edit-box.

So, it would be cool if elrsv3.lua could display it without crashing ;-)

wimalopaan, 2024-10-16:
> Just tested it with Openi6x, which simply displays a string (0x0a) parameter, but has no edit-box.

Openi6x now fully supports editing string parameters.

Maybe it would be a way to distinguish between the b&w and color radio to overcome possible memory limitations?

pfeerick, 2024-10-19:
> Well, maybe it could be integrated into EdgeTx as C++, e.g. like Openi6x this has done?

OpenI6X/opentx has done that because it was not possible to configure ELRS on handset at all otherwise, since there is no SD card on that handset. I see basically zero chance of that happening with EdgeTX, as we won't want to have to take on the additional workload of syncing with ELRS everytime they change something, and they won't want to have to keep doing PRs against EdgeTX to get things to change. 

IMO, ELRS probably should consider splitting the file soon, as there is a lot of stuff that could be done nicer for colorlcd and touch, rather than keeping one file for both UIs, but that is a decision they need to make, when it makes sense for them to do so. After all, I'm not the one is maintaining and providing support for this stuff ;)

wimalopaan, 2024-10-19:
> IMO, ELRS probably should consider splitting the file soon, as there is a lot of stuff that could be done nicer for colorlcd and touch,

Yes, Lvgl would be real cool.