ExpressLRS/ExpressLRS #2864: Bind Passphrase removed when configure Home Wifi

sokil, 2024-07-28:
\## Current Behavior

When home wifi configured or configuration forget in wifi ui, binding passphrase removed from config.

## Steps to Reproduce

1. Configure binding phrase in Web UI or during flash
2. Open Web UI and go to WiFi tab
3. Configure Home WiFi or forget configured
4. Power Off TX
5. Power On TX
6. Binding phrase removed

Storing of passphrase was removed in commit f701140ea39acdd5abf43019fea349b2da1a5fb9 in file lib/OPTIONS/options.cpp

```
 // hardware_init prototype here as it is called by options_init()
@@ -202,11 +209,6 @@ void saveOptions(Stream &stream, bool customised)
 {
     DynamicJsonDocument doc(1024);
 
-    if (firmwareOptions.hasUID)
-    {
-        JsonArray uid = doc.createNestedArray("uid");
-        copyArray(firmwareOptions.uid, sizeof(firmwareOptions.uid), uid);
-    }
     if (firmwareOptions.wifi_auto_on_interval != -1)
```

So any call of method saveOptions will remove passphrase from stored config /options.json.

Affected functionality:
* Store Home WiFi Credentials
* Forget stored WiFi Credentials

Currently passphrase may be updated only by web ui in file lib/WIFI/devWIFI.cpp:

```
static void UpdateSettings(AsyncWebServerRequest *request, JsonVariant &json)
{
  if (firmwareOptions.flash_discriminator != json["flash-discriminator"].as<uint32_t>()) {
    request->send(409, "text/plain", "Mismatched device identifier, refresh the page and try again.");
    return;
  }

  File file = SPIFFS.open("/options.json", "w");
  serializeJson(json, file);
  request->send(200);
}
```

## Possible Solution (Not obligatory)

Allow method saveOptions to store binding passphrase ir implement method that can update options consistently without remowing important things.

## Your Environment

- TX hardware: any
- ExpressLRS version (TX & RX MUST MATCH): 3.4.*

CapnBry, 2024-07-29:
I appreciate your thorough description and analysis of this situation. You're correct that the UID is not copied over to spiffs if the config is modified in any way, however I am unable to reproduce any error because of it. Once the UID has been copied to config, it no longer needs the UID stored in either flash location (appended to the firmware or in spiffs) so it isn't important information.

Here's what I tried (3.4.3):
* Configure binding phase by flashing. Confirmed the receiver is bound.
* Send RX to wifi mode, change from home network to AP mode. Reboot.
* Receiver is still bound with the full UID.

Also tried:
* Flashing with no home network but with binding phrase, waiting for it to go to AP mode, then setting the home network via the webui.
* Flashing with home network, but no bind phrase and setting it via the webui. Reboot. Is bound. Go to wifi again. Clear home network. Reboot. Still bound.

This could affect loanable mode in 3.5 but everything seems to be working in 3.4

CapnBry, 2024-07-29:
Thought I had found a corner case that could replicate it but
* Flash without bind phrase, with home network
* From the webui, enter a bind phrase, save (no reboot)
* Clear home network, save, reboot
* Receiver is bound

Also tried with no home network and adding home network after bind phrase.

CapnBry, 2024-07-29:
Oh dammit I just noticed your report is about the TX not an RX. Well, I've used all my time for today so I'll have to get back to this.

sokil, 2024-07-29:
Sorry, i had to point about TX more clearly :(

CapnBry, 2024-07-30:
No problem! I appreciate the effort you put into this and made my job easy.

PR #2870 (2024-07-30): Unremove firmwareOptions uid when customizing
Returns some code removed during #2542 which I thought was unneeded. When saving a customized json firmwareOptions to SPIFFS, the binding UID was not copied. This led to TX modules forgetting their bindphrase if any firmwareOptions is customized.

It was unneeded on the receiver since the config storage copies the value out immediately, so the json is only needed to transfer the UID from the flashing process to config storage. However, I forgot the TX does not store the UID in config storage so the JSON must always contain the UID.

The RX's Bindin [... cut]
closed 2024-07-30 by JyeSmith