ExpressLRS/ExpressLRS #2880: telemetry issue with TX16S radio and Express LRS 2.4  only 10 sensors

ermanpehlivan, 2024-08-08:
### Discussed in https://github.com/ExpressLRS/ExpressLRS/discussions/1423

<div type='discussions-op-text'>

<sup>Originally posted by **canalrodrigo** February 10, 2022</sup>
With TX16S radio and Express LRS 2.4 , I can only get it to find 10 sensors: (1RSS, 2RSS, RQly, RSNR, ANT, RFMD, TPWR, TRSS, TQly, TSNR) I already tried updating the RX and TX to the latest firmware, with the telemetry option, I also deleted all sensor and discover new sensor, and also increase the Packet Rates but still don’t get the rest of the sensors </div>

Hi guys, 

I have the same TX16S Radio and RM Express LRS 2.4 RP1, I can get it to find only 10 sensors. Because my flight controller [(Xilo Stax F4)](https://www.manualslib.com/manual/1576175/Xilo-Stax-F4-Flight-Controller.html) has only 2 UART solder pads to use. But I have a DJI o3 Unit, GPS, and the Express LRS. 

I connect the DJI o3 Unit to the UART3.
I connect the GPS with Compass to the UART4.

So, I do not have any UARTS to connect Express LRS except SBUS. I solder the wires correctly, everything is working. Binding, talking to the FC. I also set up the RP1's protocol to Inverted CRSF to be able to communicate with each other.

When I do this, I get only 10 Sensors. But when I change the CRSF instead of INVERTED, I get more than 20 sensors. Is there any solution for me?

ermanpehlivan, 2024-08-08:
I also tried to add Soft Serail 1 for the Telemetry option to Smartport, but still the same. I got only 10 sensors.

JyeSmith, 2024-08-16:
This is probably because the sbus pad requires inverted, while the tlm output from the FC in non inverted.

So you can either have inverted crsf which will give you RC input to the FC, and only the 10 sensors from the Rx.

Or non inverted, and RC will not work but you get all sensors from the FC.

The Rx can not do inverted RC and non inverted tlm input.  

Your other option is to find the non inverted input to the FC for the sbus pad.  This is a FC manufacturer question.