ExpressLRS/ExpressLRS #2889: Telemetry connected but no receiver binding in INAV

sisari, 2024-08-17:
- TX hardware: Radiomaster Ranger Nano TX ;  Jumper AION Nano T-Pro TX
- RX hardware: Radiomaster R4P-TD diversity receiver
- Handset model: Jumper T-Pro v2 with external TX module
- OpenTX version : latest EdgeTX 2.10.2  Centurion
- ExpressLRS version (TX & RX MUST MATCH): 3.4.3 
- Packet Rate: 150Hz
- Telemetry Ratio: 1:32
Is it software or hardware ? How te debug issue ?
Exaclty the same above modules on  Jumper T-Lite v2 work perfectly !!

JyeSmith, 2024-08-17:
Sounds like a possible setup issue with the handset.  Discord will be the best place for user support.

sisari, 2024-08-17:
closing issue - found root cause: changing TX modules causes EdgeTX to clear INPUTS and MIXES sections !!

closed 2024-08-17 by pkendall64

mha1, 2024-08-17:
> closing issue - found root cause: changing TX modules causes EdgeTX to clear INPUTS and MIXES sections !!

No way