ExpressLRS/ExpressLRS #2892: Wifi is enabled after 60 seconds despite unchecking auto_wifi_on_interval

LeasyBen, 2024-08-21:
Flashing the Happymodel EP1 through BFpassthrough, despite unselecting the auto_wifi_on_inverval definition the wifi comes on after 60s making it impossible to connect to the transmitter without a power cycle. Trying to flash 3.4.3.

Also, for some reason the flash fails when "erase before flash" is selected but unchecking that seems to be fine.

pkendall64, 2024-08-21:
Unchecking wifi on interval does NOT mean that it's disabled, it means it uses the default.

LeasyBen, 2024-08-21:
> Unchecking wifi on interval does NOT mean that it's disabled, it means it uses the default.

On previous versions, unchecking this box seemed to disable wifi. What can I set the value to now in order to effectively disable this feature?

gsezz, 2024-08-25:
You can set the interval to a very high timespan. Like 2147483647 seconds, which is about 70 years.

But it would be nice if there was an more intuitive option to completely disable Wi-Fi mode and keep the Rx available for TX Connections.

pkendall64, 2024-08-25:
There is, it's a bug which has a fix ready for the 3.5 release.
You should be able to turn the checkbox off and it should disable the wifi auto-on completely.

anti-vaxxer, 2024-09-02:
duplicate of #2699

PR #2897 (2024-08-26): Fix disabling of wifi-auto-on-internal
When the wifi-auto-on checkbox is disabled in Configurator, no value is given to the options JSON file, this is correct.
As part of the updated binding PR #2542 it changed the default (missing) value to 60 seconds rather than -1 (disabled).
The binding PR also broke the web UI when an empty field for the wifi-auto-on sent no value, so the 60 seconds kicked in rather than the -1!

This PR just reinstates the -1 behaviour as the default value for the missing value case.

Fixes #2892
closed 2024-09-22 by pkendall64