ExpressLRS/ExpressLRS #2896: ELRS Lua unresponsive in 3.4.3

MrMinecraft97, 2024-08-22:
## Current Behavior

Lua script V3 on my unmodded QX7 does not load at 115200 and loads very very slow on 400k. Works on 3.1.1, but not on 3.4.3. Similar to issue #2010.

## Steps to Reproduce

1. Updated from 3.3.1 on module to 3.4.3

2. updated to newest lua script from configurator

3. script no longer functions correctly on tx

4. reverted back to 3.3.1 on module and scripts loads, slowly, but it loads, much faster than updated version.

## Details

Taranis QX7 (no crossfire mod) running OTX 2.3.15-otx. Issue occurs at 115200 baud and 400k buad. Does not load at 115200, loads very very slow at 400k.

Unable to use lua script to any reasonable capacity.
Standard update from wifi with newest version of configurator on mac.

**Works fine on 3.3.x, but issues arrive on 3.4.x.**

## Your Environment

- TX hardware: HappyModel ES2400 TX, JR bay
- RX hardware: N/A
- Handset model: QX7, non crossfire mod
- OpenTX version 2.3.15-otx (1004ef7c)
- ExpressLRS version (TX & RX MUST MATCH): 3.4.3, works on 3.3.1
- Packet Rate: Slow on 400k, doesn't load at 115200
- Telemetry Ratio: STD
- user_defines: ISM_2400, UART_INVERT, SSID, Password, wifi on 60sec

mha1, 2024-09-07:
Increase your baud rate to 921k or higer

mha1, 2024-09-26:
got any feedback?

MrMinecraft97, 2024-09-26:
> got any feedback?

Unfortunately my radio (Taranis QX7) cannot do higher than 115200 without the inverter mod. I don’t currently have access to one of those, so I can’t speak to it. My other QX7 has the mod, and it works alright. Just can’t do the higher rates that 500hz.

mha1, 2024-09-26:
I really can't see a problem then. Higher packet rates require higher baud rates. Time to retire the QX7.

MrMinecraft97, 2024-09-26:
> I really can't see a problem then. Higher packet rates require higher baud rates. Time to retire the QX7.

I do remember using 2.x and earlier 3.x versions with the modded QX7 before I modded it and it worked flawlessly until like 250hz. With this version of the script and the unmodded radio, the script won’t even load.

pkendall64, 2024-09-26:
Yes, 250Hz used to work on 115200, but a platform update for fix some other issues meant that 250Hz would not work anymore on 115200. So we had to lower the max packet rate for lower baud rates. Some work was done to optimise the pipeline in 3.4 so an internal module can do 250Hz on 115200, but an external module is limited to 200Hz.
If you are interested you can read the PR the did some of the optimisations #2772

MrMinecraft97, 2024-09-26:
That makes sense. The part that I'm more concerned about however, is that the script won't load at all at 115200. I'm fine with not being able to do 250hz or higher, but it is basically unusable even for the lower packet rates if the script doesn't load. Maybe like a compatibility mode that allows the script to run with lower throughput to the transmitter module? Not sure if that is even possible. Thanks

pkendall64, 2024-09-26:
Yeah, theres not a lot we can do about that. The QX7 (actually all Frsky radios) use a slow inverter and this causes the issues for a lot of users. If possible you really should do the crossfire mod.

MrMinecraft97, 2024-09-26:
That's what I plan on, just not going to spend $6 on shipping for a $1.99 part. Wish they could just throw it in an envelope.

pkendall64, 2024-09-26:
If only Frsky had have used the $0.05 part rather than the $0.04 part 😆

mha1, 2024-10-09:
can we close this?

MrMinecraft97, 2024-10-09:
I'd like to be fixed if possible, but if not, then go ahead.

mha1, 2024-10-09:
> Yeah, theres not a lot we can do about that. The QX7 (actually all Frsky radios) use a slow inverter and this causes the issues for a lot of users. If possible you really should do the crossfire mod.

@MrMinecraft97 It can only be fixed on your side

closed 2024-10-09 by mha1