ExpressLRS/ExpressLRS #2905: Signal lost, failsafe with no signal recovery with PWM receiver (EU-LBT)

mustangp51d1, 2024-08-29:
## Current Behavior

I was flying my RC glider airplane using a RadioMaster ER6GV PWM receiver (with CE_LBT FW) in the airplane and at low altitude (15-20m) and at short distance (30-40m) suddenly I received the 'Telemetry lost' warning message. It did not only lose the telemetry information but the signal as well, the receiver went into failsafe. Signal did not recover neither in the air nor while walking closer to the RC model on the ground.
I am using this EU LBT receiver for a short period of time, this was the first issue.

Also I use the same system with another, but no CE_LBT receiver (with 3.3.0 ISM2G4 FW) for almost a year without any similar issue.

I read similar cases here at Github from the previous year. Since then there are some new ELRS releases and I would like to know which one would solve my issue. I want to avoid such issues in the future.

Thank you in advance for your help.

## Steps to Reproduce

I was not trying to reproduce this problem.

## Possible Solution (Not obligatory)

I hope that one of the recent ELRS version will be the solution, but I would like to be sure what should I do. 

## Details

Please see details above.

## Your Environment

- TX hardware: HappyModel ES24TX Slim PRO ELRS
- RX hardware: Radiomaster ER6GV EU-LBT
- Handset model: Radiomaster Zorro with built-in 4-in-1 module and external ELRS module
- OpenTX version (including nightly number) EdgeTX 2.8.4
- ExpressLRS version (TX & RX MUST MATCH): TX 3.0.1, RX LBT_3.3.1 (CE_LBT)
- Packet Rate: 333 Hz Full
- Telemetry Ratio: 1:128
- Power output: Dyn Max 250 mW

madrulzzz, 2024-08-29:
I had the same issue on the previous version. Just update to 3.4.0. There are later versions too, but this is the one I extensively tested in the past 3 months.

mustangp51d1, 2024-08-29:
> I had the same issue on the previous version. Just update to 3.4.0. There are later versions too, but this is the one I extensively tested in the past 3 months.

Thank you! When you had the issue, did you you manage to reproduce the issue before updating to 3.4.0 or were you not testing your system after the issue?

I checked and in the FW versions after 3.4.0 there are not really further EU LBT related bug fixes.

mha1, 2024-08-29:
3.4.2 had two LBT fixes

madrulzzz, 2024-08-29:
> Thank you! When you had the issue, did you you manage to reproduce the issue before updating to 3.4.0 or were you not testing your system after the issue?
>
> I checked and in the FW versions after [...]

I faced the issue while flying and crashed. Like you said, it didn't connect back even after I walked next to the model. Back on the bench I couldn't reproduce it, but I started checking in GitHub and found that this was an identified bug with EU LBT version. I followed the issue on GitHub and saw that it was fixed in 3.4.0

There are more fixes for EU LBT in versions after 3.4.0, however they are more edge case scenarios and the link had been rock solid for me in the past 3 months. I will now update only if there is any new feature that I need that gets introduced in a later build. (More on the lines of don't fix what's not broken)

But if you're comfortable you should update to the latest version.

mustangp51d1, 2024-08-29:
> 3.4.2 had two LBT fixes

Thanks for this information, I did not see them in the description.

mustangp51d1, 2024-08-29:
> I faced the issue while flying and crashed. Like you said, it didn't connect back even after I walked next to the model. Back on the bench I couldn't reproduce it, but I started checking in GitHub [...]

Thank you! It is good to hear about your experience in the past 3 months.

mustangp51d1, 2024-08-31:
> Thank you! It is good to hear about your experience in the past 3 months.

One more question just to be sure: You are using 3.4.0 EU LBT firmware both on the TX module and on your receiver as well, right?

Thank you!

madrulzzz, 2024-08-31:
> One more question just to be sure: You are using 3.4.0 EU LBT firmware both on the TX module and on your receiver as well, right?
>
> Thank you!

Yes, it's good practice to match the versions on both.

mha1, 2024-09-02:
Bottom line: a considerable amount of work, fixes, improvements, and testing went into LBT for 3.4. Use 3.4.3.

Can we close this issue?

madrulzzz, 2024-09-02:
> Bottom line: a considerable amount of work, fixes, improvements, and testing went into LBT for 3.4. Use 3.4.3.
> 
> Can we close this issue?

100%, this is a resolved issue. We should close it, so that the team can focus on current issues.

mustangp51d1, 2024-09-02:
> > Bottom line: a considerable amount of work, fixes, improvements, and testing went into LBT for 3.4. Use 3.4.3.
> > Can we close this issue?
> 
> 100%, this is a resolved issue. We should close it, so that the team can focus on current issues.

Yes, this issue can be closed. Thank you for your help!

closed 2024-09-02 by mustangp51d1

Moumouls, 2024-12-30:
Hi, I encountered an issue today with version 3.3.X. I've been flying with 333Hz EU_LBT for a long time.

For the first time, I flew on the beach with a hydroplane. Yesterday, I lost signal and never recovered it (smooth crash in the sea thanks to the gyro's failsafe). I wasn't sure what happened.

Today, I flew again without any incidents in the air. However, while the model was still powered on the ground, I traveled 50m, and suddenly the connection was lost and never recovered. The LED was just blinking yellow.

This felt really weird since I've never had an issue like this with many planes and ExpressLRS.

I noticed one thing: the first binding time at 333Hz is really long, around 10-15 seconds, but at 100Hz, it's almost instant.

I switched back to 100Hz, as I understand it could be more resilient in some cases. For a hobby fixed-wing plane, I don't need 333Hz. I also updated the TX and RX to 3.4.3 for the next flight.

I hope this helps. Maybe the issue should be reopened. I don't see a linked PR. Is it possible to link the PR related to this issue?

mha1, 2024-12-31:
> I noticed one thing: the first binding time at 333Hz is really long, around 10-15 seconds, but at 100Hz, it's almost instant.

To avoid this with 3.5: https://www.expresslrs.org/info/init-rate/

3.4 will do the same if you don't set the init rate using the LUA script receiver menu

Moumouls, 2025-01-03:
@mha1 oh i was just curious about the PR fixing this issue to check what was wrong ! If you have the related PR for the full connectivity lost issue i'will be very interested ! I never had issue with ELRS and then it happened 2 times on the beach (first time i use the plane on the beach), it's just to check on my side if it was super rare software issue or if the beach environment have something special, and i should avoid some settings to avoid connectivity issue, resulting in expensive crashes !

mha1, 2025-01-04:
> @mha1 oh i was just curious about the PR fixing this issue to check what was wrong ! If you have the related PR for the full connectivity lost issue i'will be very interested ! I never had issue [...]

There's no single PR. 3.4.0 had a number of PRs that improved the inner workings of LBT (timing, sequence of events, code improvements etc). 3.4.3 added two PRs to address corner cases.

Moumouls, 2025-01-04:
okay, thank you for taking the time to reply. @mha1